habuild/hassio-addons #13: HAS thinks haos is running while haos says it is not running.

FritsOV, 2022-07-19:
in Home Assistant i thought haos-sbfspot stoped working. so i did numerous tries to 
solve the problem. 
But after a reinstall i still got the same error: the logfile supervisor:

22-07-20 00:31:56 WARNING (MainThread) [supervisor.addons.addon] a51a23d8_haos-sbfspot is already running!

in Add on -> haos-sbfspot  is the red dot in the upper corner. there is no restart choice only the choice to start sbfspot. 
For some odd reason HAS thinks that the programm is running while the programm itself says it is not running. 
i can see that it didn't give any output to pvoutput,org  last update time was  15:15

habuild, 2022-07-20:
Hi fritzOV,

Tropfen had a similar issue.
https://community.home-assistant.io/t/sbfspot-bluetooth-homeassistant-addon/402671/62

It sounds like the container isn't being removed properly when it's uninstalled.

I will have to investigate, why that is happening.

Can you detail your system a bit? Is it HAOS or HA supervised? HA Versions, Pi or nuc etc..

If you can get root/host access in HAOS, you can probably remove the old container with docker or portainer.

habuild, 2022-07-20:
So I just tried to remove the addon on an up to date pi3 64bit HAOS device and it worked without issue. 

```
22-07-20 14:47:32 INFO (SyncWorker_6) [supervisor.docker.interface] Stopping addon_a51a23d8_haos-sbfspot application
22-07-20 14:47:33 INFO (SyncWorker_6) [supervisor.docker.interface] Cleaning addon_a51a23d8_haos-sbfspot application
22-07-20 14:47:33 INFO (SyncWorker_6) [supervisor.docker.interface] Removing image ghcr.io/habuild/aarch64-addon-haos-sbfspot with latest and 2022.3.5
22-07-20 14:47:34 INFO (MainThread) [supervisor.addons.addon] Removing add-on data folder /data/addons/data/a51a23d8_haos-sbfspot
22-07-20 14:47:34 INFO (MainThread) [supervisor.addons] Add-on 'a51a23d8_haos-sbfspot' successfully removed
```

FritsOV, 2022-07-20:
> Hi fritzOV,
> 
> Tropfen had a similar issue. https://community.home-assistant.io/t/sbfspot-bluetooth-homeassistant-addon/402671/62
> 
> It sounds like the container isn't being removed properly when it's uninstalled.
> 
> I will have to investigate, why that is happening.
> 
> Can you detail your system a bit? Is it HAOS or HA supervised? HA Versions, Pi or nuc etc..
> 
> If you can get root/host access in HAOS, you can probably remove the old container with docker or portainer.
> 
> I just tried a reinstall on HA supervised and it worked correctly(it removed the container), so it may be an issue with HAOS.

Hello HAos,

the system is a  Pi 3b+ 
I am not that familiar with what you all ask but i give it a try.
I stoped the service  manualy under services but that didn't do the trick.
Its not only haos-sbfspot that stoped working, also Resol Deltasol sensors are not showing any data. 
My first guess is the automatic update from the add on SSH  becaus pvoutput.org stoped getting 
data while this is about the time that SSH automatically got an update.  It gets data....

## System Health

version | core-2022.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.32-v8
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.26.0
Stage | running
Available Repositories | 1071
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 29.0 GB
disk_used | 5.3 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Samba share (10.0.0), SSH & Web Terminal (11.0.2), Mosquitto broker (6.1.2), MariaDB (2.5.0), phpMyAdmin (0.8.0), Duck DNS (1.15.0), HAOS-SBFspot (2022.3.5), ESPHome (2022.6.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 7
views | 9
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 11 juli 2022 om 09:28
-- | --
current_recorder_run | 20 juli 2022 om 12:43
estimated_db_size | 58.76 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

I also see that the sensors from my sma do give data. 
<html>
<body>

sensor.sma_daily_energySMA Daily Energy | 2.175
-- | --

</body>
</html>

[sensor.sma_daily_energy](https://XXXXXXXX.duckdns.org/developer-tools/state#)
SMA Daily Energy
2.175

From the log files under supervisor:

After a reboot:

22-07-20 13:14:20 INFO (SyncWorker_5) [supervisor.docker.interface] Restarting ghcr.io/home-assistant/raspberrypi3-64-homeassistant
22-07-20 13:14:56 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
22-07-20 13:15:18 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-07-20 13:15:19 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-07-20 13:15:21 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-07-20 13:15:21 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-07-20 13:15:21 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-07-20 13:15:21 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
22-07-20 13:15:21 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/habuild/hassio-addons repository
22-07-20 13:15:27 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
22-07-20 13:15:27 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
22-07-20 13:16:07 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance

**HEO SBfspot:**
Reading config '/etc/sbfspot/SBFspot.cfg'
Tue Jul 19 17:45:00 2022: INFO: Starting...
[SBFspot Upload Log Latest]
[16:53:30.935] WARNING:XXXXXXXXXX is not yet member of SBFspot Team. Consider joining at http://pvoutput.org/listteam.jsp?tid=XXXX

this is strange because i have an account on pvoutput.org and going to that adres you will see directly my output. 
this warning was always there so straight after installation and working system. 

**MQQT Broker**
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] mosquitto.sh: executing... 
[12:48:48] INFO: Setting up user XXXXXXXXXX
[12:48:51] INFO: Certificates found: SSL is available
[cont-init.d] mosquitto.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[12:48:53] INFO: Starting NGINX for authentication handling...
[12:48:54] INFO: Starting mosquitto MQTT broker...
1658314134: Warning: Mosquitto should not be run as root/administrator.
[12:48:58] INFO: Successfully send discovery information to Home Assistant.
[12:48:59] INFO: Successfully send service information to the Supervisor.

FritsOV, 2022-07-20:
did what you wrote.
1. uninstalled HAOS-SBFspot
2. installed portainer protection mode off
3. could not stop/kill restart/pause or remove the container. 
All come back with failure status code 500

In images i can't remove the image of HAOS with the failure:  unable to remove containter. 
the first time i tried it gave the failure: used by another b588e997382c
In the listing of contianer HAOS is still visible after uninstall with status running.

I thougth this is mariaDb so i uninstalled in HAS, 
Tried again but cant remove the container also... 
Even stoped the MQQT broker but no luck.

habuild, 2022-07-21:
Status code 500 seems to be a locked file issue.. Which is basically the problem. I am not sure what is locking the file.

Might be best to do a partial restore of a backup, excluding HAOS-SBFspot.

Basically you need to able force stop the application while home assistant is stopped, which is abit difficult from a HAOS system.

A restore is probably the best way forward.

FritsOV, 2022-07-21:
Hello, 

Today i did a clean setup and it looks all works except for one thing. i dont get sensors for the SMA
In Yaml file i have configured 

mqtt:
    sensor:
        
      - name: "SMA Sunny Boy"
        state_topic: "homeassistant/sbfspot_SMA_XXXXXXX/sbfspot_110XXXXXX"
        value_template: '{{ value_json.InvName }}'
        icon: "mdi:flash"
etc...

it updates pvoutput.org and messages are sent with Mqtt

habuild, 2022-07-21:
https://github.com/habuild/hassio-addons/blob/main/.images/sbfspot_sensors.yaml

mqtt and sensor need an indent like below. 

```
mqtt:
    sensor:
        
      - name: "SMA Sunny Boy"
        state_topic: "homeassistant/sbfspot_Inverter_name/sbfspot_Serial_no"
        value_template: '{{ value_json.InvName }}'
        icon: "mdi:flash"
```

habuild, 2022-07-26:
Unknown cause of file locking, possibly caused by update of mariadb. Resolving file lock by partial restoration of HA, while skipping HAOS-sbfspot and then reinstalling HAOS-Sbfspot fixes the issue.

closed 2022-07-26 by habuild