habuild/hassio-addons #20: Make MySQL and PVOutput optional

virusmoere, 2022-07-30:
Please make SQL and PVOutput optional since not everyone is using it.

habuild, 2022-07-30:
Can you explain your usecase further please?
Also what device are you using. pi4 64bit?

Are you only using mqtt? You want it to not use SBFspot upload at all and send no data to a database?

**options available atm.**
1. You can put in fake info SQL hostname and PVoutput details in that case. It will show errors in the log, but SBFspot will still 
run on it's 5 minute cycle I thnk. I can't confirm until morning.

2. You are welcome to fork the repo. It is probably easier to change the repo details to manage yourself and delete the crontabs in the Dockerfile. 

    If you delete line 75 and 76 SBFspotUpload won't start and taillog.sh won't run as it won't be needed.

    You can also then change lines 22-32 in the Dockerfile to make SBFspot without DB access (make nosql). 
  Change line 28 to make nosql, Then delete lines 29-32 so that SBFspotUpload isn't compiled
  
    https://github.com/SBFspot/SBFspot/blob/master/SBFspot/makefile
  **Compilation: SBFspot**
  **make nosql|sqlite|mysql|mariadb**
  
    https://github.com/SBFspot/SBFspot/blob/master/SBFspotUploadDaemon/makefile
  **Compilation: SBFspotUpload**
  **make sqlite|mysql|mariadb**

3. Total rewrite of the dockerfile to use a precompiled release in a new addon repo
https://github.com/SBFspot/SBFspot/releases/tag/V3.9.5

4. Figure out how to write bashio scripting to make SQL and PVOutput Optional. 

Option 1 is my preferred option. I could give you hand with option 2-3
Option 4 is unlikely to happen anytime in the near future (bashio has limited documentation that I can find)

virusmoere, 2022-07-30:
Hi,

I am using Home Assistant OS on a proxmox VM. I was not able to bring this addon in a working state because it is not detecting my Bluetooth adapter yet.

Currently I have SBFspot running on a dedicated VM and I am using the CSV export (sunwatcher app on Android) and MQTT output for Home Assistant only.

habuild, 2022-07-30:
Ahh Bluetooth hardware issue. Do you know what architecture your add-ons use, amd64 I assume? I haven't found a solution for VMs or amd64 yet, seems to be common issue. So at this point there would be no benefit to making a addon without SQL or PVoutput. 

I'm not familiar with sunwatcher, it wasn't in the playstore. Only available as apk. Do you run it on a separate tablet? I assume it just reads your csv files. I haven't actually tested this addon with csv files, I didn't want it writing lots of csv files, hence the database. 

Is

virusmoere, 2022-07-31:
Yes I am using an amd64 architecture on the VM.

Sunwatcher is reading the CSV files originally from the sunny explorer tool. I serve the files via a separate webserver. I will the the options above if I got the Bluetooth stuff working 👍

woutercoppens, 2022-08-05:
I second to make MySQL and PVoutput optional. I want to use this addon to retrieve the data from an old inverter and store it in the energy dashboard of HA.