habuild/hassio-addons #21: Ethernet connected solar inverter

woutercoppens, 2022-08-05:
Does this plugin work with an ethernet connected inverter? 
I use the SolarLog 100e and it's a rather old inverter.

habuild, 2022-08-05:
It's not set up to use the Ethernet connection. Technically possible to do, although I have no way of testing if that works. It's just not available in the config section. 

Do you currently use SBFspot to connect to your router?

habuild, 2022-08-08:
https://community.home-assistant.io/t/sbfspot-bluetooth-homeassistant-addon/402671/132?u=hasqt

Try this one. No idea if it works I don't have ethernet router.

woutercoppens, 2022-08-08:
I looked into SBFspot but my inverter is not speedwire compatible. Sorry.

habuild, 2022-08-08:
I was under the impression SBFspot uses both speedwire or Ethernet. Although like I say I only have Bluetooth so I'm not actually sure.

dweston, 2023-02-07:
Hi @habuild , I have a Sunny Boy SMA 4000TL-21 that I have been successfully using SBFspot with to extract data from the inverter and upload it to PVoutput. SBFspot is installed on a separate Rpi to my HA Rpi, and is connected via Ethernet to a Speedwire/Webconnect Data Module in the inverter.

I've been wanting to integrate this data directly into HA for some time, instead of having to access it via the Web. So I was delighted to have come across this 'HAOS-SBFspot' plug-in.

Only issue is the Ethernet configuration. With my most recent set-up of SBFspot I've relied on the Wizard script to create the config file for me. Reading the 'HAOS-SBFspot' docs once the plug-in is installed, it appears I simply follow the same path to input the inverter IP address and credentials, leaving the Bluetooth set-up fields blank. Am I reading that correctly?