habuild/hassio-addons #32: Bluetooth can't connect with external USB dongle

mcc05, 2022-08-30:
I have an external USB dongle on my HA, using stock image on an IntelNUC
Home Assistant 2022.8.7
Supervisor 2022.08.5
Operating System 8.5
Frontend 20220802.0 - latest

The device is recognised in HAOS as a bluetooth radio etc and is working fine.

From within a shell inside your docker container its is also recognised,

_bash-5.1# bluetoothctl
Agent registered
[CHG] Controller **00:E0:4C:AA:47:03** Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:E0:4C:AA:47:03 Discovering: yes
[NEW] Device 48:74:12:91:34:F4 OnePlus Nord 2T 5G
[NEW] Device 68:84:FA:AC:F0:3F 68-84-FA-AC-F0-3F
**[NEW] Device 00:80:25:23:9D:9E BlueCN+00A D-239D9E**_

And my inverter is found above

But when SBFSpot runs it fails to connect to the inverter every time like:

_[16:42:14] INFO: 
[16:42:14] INFO:  Linux 5.15.60 #1 SMP Tue Aug 16 12:26:36 UTC 2022 x86_64
[16:42:14] INFO: 
[16:42:14] INFO: 
[16:42:14] INFO:  [Host Bluetooth MAC Address] Controller **00:E0:4C:AA:47:03** homeassistant [default]
[16:42:14] INFO: 
[16:42:16] INFO: 
[16:42:16] INFO: 
[16:42:16] INFO:  ||    Generating Bluetooth Configs    ||
[16:42:16] INFO:  ||    /usr/bin/sbfspot/SBFspot.cfg    ||
[16:42:16] INFO:  || /usr/bin/sbfspot/SBFspotUpload.cfg ||
[16:42:16] INFO: 
[16:42:16] INFO:  [Starting SBFspotUpload]
[16:42:16] INFO: [Starting cron in foreground]
SBFspot V3.9.5
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v -ad0 -am0 -mqtt -finq
Reading config '/usr/bin/sbfspot/SBFspot.cfg'
Tue Aug 30 16:42:46 2022: INFO: Starting...
sunrise: 06:11
sunset : 19:50
Connecting to **00:80:25:23:9D:9E** (1/10)
Connecting to 00:80:25:23:9D:9E (2/10)
Connecting to 00:80:25:23:9D:9E (3/10)
Connecting to 00:80:25:23:9D:9E (4/10)
Connecting to 00:80:25:23:9D:9E (5/10)
Connecting to 00:80:25:23:9D:9E (6/10)
Connecting to 00:80:25:23:9D:9E (7/10)
Connecting to 00:80:25:23:9D:9E (8/10)
Connecting to 00:80:25:23:9D:9E (9/10)
Connecting to 00:80:25:23:9D:9E (10/10)
Tue Aug 30 16:42:55 2022: CRITICAL: bthConnect() returned -1_

I have tried the same adaptor on a raspberry pi (internal BT disabled) with SBFspot not in docker installed and works fine.

There is not enough debugging out of SBFspot to suggest the problem.

habuild, 2022-08-30:
Good post info wise.

Yeah debugging the connection is lacking in information. I did ask over in the discussions section of SBFspot. If they could offer any further insight into that.

You could try and watch dmesg at the same time as connecting to see if anything interesting pops up. I haven't really done that myself, so not sure what you would be looking for other than BT dongle disconnects or similar.

Range, range is often the underlying issue. You didn't mention how far away both machines were. With a cheap usb2 dongle my connection starts being difficult after about 5 metres. 

My inverter gets abit picky if I manage to hit two simultaneous connections from different machines.. Doesn't sound like that is what is happening, but worth noting. 

When you scan with Bluetoothctl is it spamming messages? If it only has one device show up and isn't scrolling super fast, that's generally a bad BT setup. 

It's not required for the addon to work, however can you trust and pair with the inverter via Bluetoothctl?  What sort of rssi number are you getting? 

Do you have the HA Bluetooth integration setup? Deleting that and leaving it not setup can help. Although I usually only have trouble there when I am tryin to connect via a VM. 

You are on a bare metal install not a VM? 

If you were keen and have a spare Sd card around it might be worth trying the addon via the Pi.

mcc05, 2022-08-31:
Its a VM setup on Virtual box on Windows  on a NUC, all the above tests where done on that.

I will do some more testing as you suggest above, when I get a bit of time, for now I disconnected the bluetooth adaptor from the VM instance, and I'm running the Windows version of SBFspot to push to the MQTT of HA and that works fine (suggesting range etc on the dongle is not the issue), just a shame it can't all be in the HA env :)

habuild, 2022-08-31:
Ahh your mileages may be better than mine. 

My desktop using vmware player can't get a connection with HAOS on a vm. It can with a vm of debian/docker/supervisor and also win10. It was struggling to pair even on both types of vm. I could not get a speaker to pair either to HAOS. 

I think I upgraded the BT device firmware on the deb/supervisor/ha vm. HAOS 2022.9.x is getting a firmware driver expansion too I think..

In saying that my mb and CPU are really old, not eligible for win11 old.. So you may have better luck

habuild, 2022-09-08:
@mcc05 Any luck with 2022.09?
I can discover the BT dongle with the Bluetooth integration on HAOS. Still can't get a connection with sbfspot though. It works with deb/docker/supervisor vm.. Still fiddling.

mcc05, 2022-09-08:
@habuild sorry not had time to test yet I will try at weekend

habuild, 2022-09-15:
@mcc05 
OS9 just came out.. I was able to connect 1st attempt

mcc05, 2022-09-15:
Same here, worked first time :)

closed 2022-09-15 by mcc05