habuild/hassio-addons #39: Difference in real energy and HAS energy

FritsOV, 2022-10-05:
I have a strange problem,
I thought my pv inverter was broke but after investigating there is a difference in what HAS registers as total of a day and what 
the real total of a day is. As of writing time the total in the energy dashboard is: 
<html>
<body>

SMA Daily Energy | 0,67 kWh
-- | --

</body>
</html>

Sas you can see the real total is: EToday": 2.911,
What causes this behaviour?

MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}) "PrgVersion": "3.9.6","Plantname": "XXXXXXXXX","Timestamp": "2022-10-05T16:15:05+0200","SunRise": "2022-10-05T07:44:00+0200","SunSet": "2022-10-05T19:03:00+0200","InvSerial": 1100267652,"InvName": "SN: 1100267652","InvTime": "2022-10-05T16:15:05+0200","InvStatus": "Ok","InvSwVer": "12.12.203.R","InvClass": "Solar Inverters","InvType": "SB 3300TL HC","InvTemperature": 0.000,"InvGridRelay": "?",**"EToday": 2.911,**"ETotal": 38394.352,"GridFreq": 49.990,"PACTot": 269.000,"PAC1": 269.000,"UAC1": 240.000,"IAC1": 1.124,"OperTm": 47924.675,"FeedTm": 43755.714,"PDCTot": 0.000,"UDC1": 350.000,"UDC2": 0.000,"IDC1": 0.910,"IDC2": 0.000,"PDC1": 318.000,"PDC2": 0.000,"BTSignal": 63.922,"InvWakeupTm": "2022-10-05T16:15:04+0200","InvSleepTm": "2022-10-05T16:15:06+0200"
Client null sending CONNECT

habuild, 2022-10-05:
Hmm, that is abit weird. Is it a regular occurrence?

Do you put EToday or ETotal to the dashboard?
Is your timezone correct perhaps the dashboard is resetting at the the wrong time?

FritsOV, 2022-10-06:
Well today i'm getting nothing. 
There is zero update and suddenly there is an error in SPot:
"EToday": 3.971
*Connection error: Connection Refused: not authorised.**
**Error 1280 while publishing to MQTT Broker**

This happens while nothing has been changed!

LOGfile HEOSBFspot:
MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}) "PrgVersion": "3.9.6","Plantname": "XXXXXXXXXXX","Timestamp": "2022-10-06T12:30:08+0200","SunRise": "2022-10-06T07:46:00+0200","SunSet": "2022-10-06T19:01:00+0200","InvSerial": 1100267652,"InvName": "SN: 1100267652","InvTime": "2022-10-06T12:30:08+0200","InvStatus": "Ok","InvSwVer": "12.12.203.R","InvClass": "Solar Inverters","InvType": "SB 3300TL HC","InvTemperature": 0.000,"InvGridRelay": "?",**"EToday": 3.971,**"ETotal": 38398.827,"GridFreq": 50.020,"PACTot": 1728.000,"PAC1": 1728.000,"UAC1": 246.000,"IAC1": 7.028,"OperTm": 47931.759,"FeedTm": 43762.206,"PDCTot": 0.000,"UDC1": 203.000,"UDC2": 0.000,"IDC1": 8.880,"IDC2": 0.000,"PDC1": 1802.000,"PDC2": 0.000,"BTSignal": 65.490,"InvWakeupTm": "2022-10-06T12:30:07+0200","InvSleepTm": "2022-10-06T12:30:09+0200"
Client null sending CONNECT
**Connection error: Connection Refused: not authorised.**
Error: The connection was refused.
Client null received CONNACK (5)
MQTT: Failed te execute '/usr/bin/mosquitto_pub' mosquitto client installed?
**Error 1280 while publishing to MQTT Broker**

LOG file MQQT:
2022-10-06 11:54:53: New client connected from 172.30.32.1:56031 as 3O39wm49HCkY5l3LVUnfoX (p2, c1, k60, u'homeassistant').

and the rest is:
2022-10-06 11:55:08: New connection from 172.30.32.1:37556 on port 1883.
2022-10-06 11:55:08: Client <unknown> disconnected, not authorised.
2022-10-06 11:55:38: New connection from 172.30.32.2:44982 on port 1883.
2022-10-06 11:55:38: Client <unknown> closed its connection. 
etcetera.

MQQT n Yaml i have defined SMA dayly energy and this sensor is adressed in the energy dashboard.

I am totaly out of clues here...

habuild, 2022-10-06:
You need to add username and password to the MQTT Addon (not the Integration). User homeassistant is not an allowed username because it used internally in Home Assistant. The enforcement of that is a recent change in Home Assistant. 

You can use your HA login I think without adding a user to the broker.. but that is probably a bad idea security wise, better to put a user and password that doesn't matter so much if it gets passed around by MQTT.

You will need to adjust the user pass in HAOS-SBFspot options to match the user and pass you create in the mqtt broker.

https://my.home-assistant.io/redirect/supervisor_addon/?addon=core_mosquitto

[image]

************************************
https://github.com/home-assistant/addons/blob/master/mosquitto/DOCS.md

>Create a new user for MQTT via your Home Assistant's frontend Configuration -> Users (manage users) , (i.e. not on Mosquitto's Configuration tab). Notes:
>
> 1. This name cannot be homeassistant or addons, those are reserved usernames.

FritsOV, 2022-10-07:
Thanks for your reply!
It worked for several weeks so MQQT worked, SBF worked and Mariadb worked correctly only thing it gives wrong results. 
After an update of HAS core it went wrong. I saw there is a new update for HAS core. 
i investigated the MQQT matter and did an update to latest Core. 
Also restarted the MQQT and now it all works except for showing the right data. 
As you can see in the log the total of energy of today is: EToday": 2.830,
But in the dashboard it gives: 
<html>
<body>

SMA Daily Energy | 1,85 kWh
-- | --

</body>
</html>

In My yaml config i have a sensor under mqqt: and this sensor is provided in the energy dashboard. 
I wonder if this behaviour has anything to do with recalculating from kWh to Kw?

      - name: "SMA Daily Energy"
        state_topic: "homeassistant/sbfspot_XXXXXXXXXXXX/sbfspot_XXXXXXXXXX"
        value_template: '{{ value_json.EToday }}'
        unit_of_measurement: "kWh"
        icon: "mdi:solar-power"
        device_class: energy
        state_class: total_increasing

**LOG file SBF:**
Current Inverter Time: 2022-10-07T12:10:06+0200
Inverter Wake-Up Time: 2022-10-07T12:10:05+0200
Inverter Sleep Time  : 2022-10-07T12:10:08+0200
MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}) "PrgVersion": "3.9.6","Plantname": "XXXXXXXXXXX","Timestamp": "2022-10-07T12:10:06+0200","SunRise": "2022-10-07T07:48:00+0200","SunSet": "2022-10-07T18:59:00+0200","InvSerial": 1100267652,"InvName": "SN: 1100267652","InvTime": "2022-10-07T12:10:06+0200","InvStatus": "Ok","InvSwVer": "12.12.203.R","InvClass": "Solar Inverters","InvType": "SB 3300TL HC","InvTemperature": 0.000,"InvGridRelay": "?", **"EToday"** : 2.830,"ETotal": 38405.249,"GridFreq": 50.010,"PACTot": 940.000,"PAC1": 940.000,"UAC1": 243.000,"IAC1": 3.869,"OperTm": 47942.440,"FeedTm": 43772.471,"PDCTot": 0.000,"UDC1": 285.000,"UDC2": 0.000,"IDC1": 3.780,"IDC2": 0.000,"PDC1": 1077.000,"PDC2": 0.000,"BTSignal": 67.451,"InvWakeupTm": "2022-10-07T12:10:05+0200","InvSleepTm": "2022-10-07T12:10:08+0200"
Client null sending CONNECT
Client null received CONNACK (0)
Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_XXXXXXXXX/sbfspot_XXXXXXXXXX', ... (765 bytes))
Client null sending DISCONNECT

(PS. as you can see the BTSignal is 67 and  raspberry pi is located on first floor while inverter is located on third floor....)

habuild, 2022-10-07:
You can add these defaults to your ETotal and EToday. It will help with error values during restarts and things.
```
value_template: '{{ value_json.EToday | float(default) }}'

value_template: '{{ value_json.ETotal | float(default) }}'
```

That shouldn't be your current problem though. EToday and ETotal don't get converted from kW to kWh
```
Energy Production:
	EToday: 0.000kWh
	ETotal: 60533.851kWh
```
They are sent as kWh from SBFspot.

I usually use ETotal on the energy dashboard, that shouldn't really matter though. EToday should still be getting reset at midnight. There will be some lag in values on the energy dashboard as it takes an average over an hour.. I am not sure it should be out by a kilowatt though. The values for previous days are also incorrect or does the figure end up being correct at the end of the day?(<sub>when it has been working, they shouldn't be correct if the messages were not getting through due to unauthorized</sub>)

You could try using ETotal instead and see if it still ends up being the wrong value. 

Timezone is correct in general settings?
https://my.home-assistant.io/redirect/general/
[image]

I think your unauthorised may have possibly been related to this error. It has been popping up a bit lately for me if I restart the MQTT broker. It sometimes stops taking my users after that.. sometimes.. Which explains why it is working again without changing anything.
```
22-10-08 02:11:13 ERROR (MainThread) [supervisor.services.modules.mqtt] There is already a MQTT service in use from core_mosquitto
```

FritsOV, 2022-10-07:
I have checked the time settings and they are okay. Amsterdam + 01:00 
but then i was wondering why you pointed to the time direction and saw in my previous post the logfile.

Current Inverter Time: 2022-10-07T12:10:06+0200  <-------------
Inverter Wake-Up Time: 2022-10-07T12:10:05+0200
Inverter Sleep Time : 2022-10-07T12:10:08+0200

Dont know how HAOS is handeling time but there is 1 hour difference in HAS time and inverter time if i read the log file correctly... 

Thats wrong and i dont know how or why this setting is changed... 
I'm sure that this is not the problem. My best guess for now is that it is the broker that didn't function correctly. 

Tomorrow i will change dashboard in Etotal sensor and will check again. Also will see how i can change the systemtime of the SMA to the right time and see what happens next....

habuild, 2022-10-08:
For what its worth. My Dashboard value is about 800wH behind my latest publish from SBFspot at 10:30am. 
Although I use ETotal 

```
SMA Energy Lifetime | 3.04 kWh   <<< dashboard
"EToday": 3.882   <<< SBFspot
```

Time is quite the pain.. 
HAOS has to set it's time. 
SBFspot has to set it's time, That should be set via the options config.

The Inverter has to set it's time. 
The inverter I use "sunny explorer" to set the time and Daylight savings dates... They shifted a couple of years ago here. So they needed to be adjusted. There is also an option to use or not use Daylight savings times which is easy to skip over. Although I don't know how you usually configure your inverter or if Amsterdam has daylight savings time changes during the year.