habuild/hassio-addons #40: cfg not being created - Invalid value for 'Locale'

karmacop, 2022-10-08:
**Describe the bug**
It doesn't appear as if a config is being created. Receiving the following lines in the log:
Reading config '/usr/bin/sbfspot/SBFspot.cfg'
Invalid value for 'Locale' de-DE|en-US|fr-FR|nl-NL|it-IT|es-ES

Using the SSH login in home assistant I can go to /usr/bin/ but there is no sbfspot directory.

**To Reproduce**
Installed addon. Used the following config:
Connection_Type: Ethernet
Sensors_HA: Create
BTAddress: _0040ad82xxxx_
Password: _password_
LocalBTAddress: "!secret LocalBTAddress"
IP_Address: 192.168.1.101
Plantname: stp_5000TL_20
Latitude: "30"
Longitude: "150
Timezone: Australia/Sydney
DateTimeFormat: "'%H:%M:%S %d-%m-%y'"
SQL_Password: "!secret mariadb_pw"
MQTT_User: _username_
MQTT_Pass: _password_
MQTT_Topic: homeassistant/sbfspot_stp_5000TL_20/sbfspot__xxxxxxxx_
MQTT_Data: >-
PrgVersion,Plantname,Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvSwVer,InvClass,InvType,InvTemperature,InvGridRelay,EToday,ETotal,GridFreq,PACTot,PAC1,UAC1,IAC1,OperTm,FeedTm,PDCTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2,BTSignal,InvWakeupTm,InvSleepTm
PVoutput_SID: "!secret pvoSID"
PVoutput_Key: "!secret pvoAPIkey"
LogLevel: info

get the following log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sbfspot (no readiness notification)
s6-rc: info: service legacy-services successfully started
[10:09:04] INFO: 
[10:09:04] INFO:  Linux 5.15.61-v8 #1 SMP PREEMPT Tue Sep 13 16:06:05 UTC 2022 aarch64
[10:09:04] INFO: 
[10:09:04] INFO: 
[10:09:04] INFO:  [Host Bluetooth MAC Address] Controller E4:5F:01:xx:xx:xx homeassistant [default]
[10:09:04] INFO: 
|   INITIALISE MQTT SENSOR CREATION   |
stp_5000TL_20
core-mosquitto
_username_
homeassistant/sbfspot_stp_5000TL_20/sbfspot__xxxxxxxx_
PrgVersion,Plantname,Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvSwVer,InvClass,InvType,InvTemperature,InvGridRelay,EToday,ETotal,GridFreq,PACTot,PAC1,UAC1,IAC1,OperTm,FeedTm,PDCTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2,BTSignal,InvWakeupTm,InvSleepTm
[10:09:05] INFO: Using Existing Device info
{
  "InvSerial": _xxxxxxxx_,
  "InvName": "STP 5000TL-20 745",
  "InvClass": "Solar Inverters",
  "InvType": "STP 5000TL-20",
  "InvSwVer": "02.55.03.R"
}
xxxxxxxx
PrgVersion Plantname Timestamp SunRise SunSet InvSerial InvName InvTime InvStatus InvSwVer InvClass InvType InvTemperature InvGridRelay EToday ETotal GridFreq PACTot PAC1 UAC1 IAC1 OperTm FeedTm PDCTot UDC1 UDC2 IDC1 IDC2 PDC1 PDC2 BTSignal InvWakeupTm InvSleepTm
[10:09:06] INFO: Setting Up PrgVersion
[10:09:07] INFO: Setting Up Plantname
[10:09:07] INFO: Setting Up InvName
[10:09:07] INFO: Setting Up Timestamp
[10:09:07] INFO: Setting Up InvTime
[10:09:07] INFO: Setting Up SunRise
[10:09:07] INFO: Setting Up SunSet
[10:09:07] INFO: Setting Up InvSerial
[... 216 lines cut ...]
[21:58:48] INFO: 
[21:58:48] INFO:  ||    Generating Ethernet Configs     ||
[21:58:48] INFO:  ||    /usr/bin/sbfspot/SBFspot.cfg    ||
[21:58:48] INFO:  || /usr/bin/sbfspot/SBFspotUpload.cfg ||
[21:58:48] INFO: 
[21:58:48] INFO: 
[21:58:48] INFO:  [Starting SBFspotUpload]
[21:58:48] INFO: [Starting cron in foreground]
SBFspot V3.9.6
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v -ad0 -am0 -mqtt -finq
Reading config '/usr/bin/sbfspot/SBFspot.cfg'
Invalid value for 'Locale' de-DE|en-US|fr-FR|nl-NL|it-IT|es-ES
INFO: SBFspotUploadDaemon Version 3.0.2
Unable to open database. Check configuration.
[2022-10-08 21:59:51.892] Error: Can't open MySQL db [SBFspot] : Unknown MySQL server host 'core-mariadb' (-2)
SBFspot V3.9.6
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v -ad1 -am0 -ae0 -mqtt
Reading config '/usr/bin/sbfspot/SBFspot.cfg'
Invalid value for 'Locale' de-DE|en-US|fr-FR|nl-NL|it-IT|es-ES

karmacop, 2022-10-09:
I had a bit of a look and I added a locale line to the config and that fixed it but now it is sending a bunch of nulls to mqtt. Possibly this is an ethernet issue as it looks like it's not picking up some settings that the bluetooth version must be. See the below log:

```
[14:56:17] INFO: Setting Up PAC1
Skipping PAC2
Skipping PAC3
[14:56:17] INFO: Setting Up UAC1
Skipping UAC2
Skipping UAC3
[14:56:17] INFO: Setting Up IAC1
Skipping IAC2
Skipping IAC3
[14:56:17] INFO: Setting Up OperTm
[14:56:17] INFO: Setting Up FeedTm
[14:56:17] INFO: Setting Up PDCTot
[14:56:17] INFO: Setting Up PDC1
[14:56:17] INFO: Setting Up PDC2
Skipping PDC
[14:56:17] INFO: Setting Up UDC1
[14:56:17] INFO: Setting Up UDC2
Skipping UDC
[14:56:17] INFO: Setting Up IDC1
[14:56:17] INFO: Setting Up IDC2
Skipping IDC
[14:56:17] INFO: Setting Up GridFreq
[14:56:17] INFO: Setting Up BTSignal
Skipping BatTmpVal
Skipping BatVol
Skipping BatAmp
Skipping BatChaStt
[14:56:18] INFO: Setting Up InvWakeupTm
[14:56:18] INFO: Setting Up InvSleepTm
Skipping MeteringWOut
Skipping MeteringWIn
Skipping MeteringWTot
[14:56:23] INFO: 
[14:56:23] INFO: 
[14:56:23] INFO:  ||    Generating Ethernet Configs     ||
[14:56:23] INFO:  ||    /usr/bin/sbfspot/SBFspot.cfg    ||
[14:56:23] INFO:  || /usr/bin/sbfspot/SBFspotUpload.cfg ||
[14:56:23] INFO: 
[14:56:23] INFO: 
[14:56:23] INFO:  [Starting SBFspotUpload]
[14:56:23] INFO: [Starting cron in foreground]
SBFspot V3.9.6
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v -ad0 -am0 -mqtt -finq
Reading config '/usr/bin/sbfspot/SBFspot.cfg'
Sun Oct  9 14:56:53 2022: INFO: Starting...
sunrise: 06:29
sunset : 19:07
Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 901316616
Device IP address: 192.168.1.101 from config
Inverter replied: 192.168.1.101 -> 181:xxxxxxxx
Logon OK
SUSyID: 181 - SN: xxxxxxxx
Device Name:      STP 5000TL-20 745
Device Class:     Solar Inverters
Device Type:      STP 5000TL-20
Software Version: 02.55.03.R
Packet status: 21
SUSyID: 181 - SN: xxxxxxxx
Device Status:      Ok
SUSyID: 181 - SN: xxxxxxxx
Device Temperature: 64.1°C
SUSyID: 181 - SN: xxxxxxxx
GridRelay Status:      Closed
SUSyID: 181 - SN: xxxxxxxx
Energy Production:
	EToday: 14.670kWh
	ETotal: 60876.550kWh
	Operation Time: 35078.00h
	Feed-In Time  : 34288.92h
SUSyID: 181 - SN: xxxxxxxx
DC Spot Data:
	MPPT 1 Pdc:   2.389kW - Udc: 393.38V - Idc:  6.076A
	MPPT 2 Pdc:   1.230kW - Udc: 173.70V - Idc:  7.089A
	Calculated Total Pdc:   3.619kW
AC Spot Data:
	Phase 1 Pac :   1.199kW - Uac: 242.48V - Iac:  4.945A
	Phase 2 Pac :   1.186kW - Uac: 240.28V - Iac:  4.937A
	Phase 3 Pac :   1.190kW - Uac: 241.50V - Iac:  4.929A
	Total Pac   :   3.575kW - Calculated Pac:   3.575kW
	Efficiency  :   98.78%
SUSyID: 181 - SN: xxxxxxxx
Grid Freq. : 49.96Hz
SUSyID: 181 - SN: xxxxxxxx
Current Inverter Time: 2022-10-09T14:56:56+1100
Inverter Wake-Up Time: 2022-10-09T06:47:24+1100
Inverter Sleep Time  : 2022-10-09T14:57:00+1100
[2022-10-09 14:56:57.892] Error: Can't open MySQL db [SBFspot] : Unknown MySQL server host 'core-mariadb' (-2)
MQTT: Publishing (homeassistant/sbfspot_stp_5000TL_20/sbfspot_xxxxxxxx) null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null
Client null sending CONNECT
Client null received CONNACK (0)
Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_stp_5000TL_20/sbfspot_xxxxxxxx', ... (166 bytes))
Client null sending DISCONNECT
Reading events: 2022-Oct-01
Sun Oct  9 14:56:57 2022: INFO: Done.
INFO: SBFspotUploadDaemon Version 3.0.2
```

habuild, 2022-10-09:
`docker exec -it addon_a51a23d8_haos-sbfspot /bin/bash`
is the command to get into the container via a console. 
This won't work in the official terminal addon however as the docker command is not available in that addon.
It should work in the community SSH and Terminal addon or a putty console.
Portainer is a much easier way to connect to the addon, if you feel the need in the future.

The above should hopefully clear up why you didn't find what you expected when using SSH
> Using the SSH login in home assistant I can go to /usr/bin/ but there is no sbfspot directory.
************
To clarify you are actually intending to use an ethernet  and not a Bluetooth connection?

You may be the first to try it with ethernet I believe.. So my side of it may be abit buggy.. the SBFspot side should work fine.
That's my disclaimer out of the way..

In the options, for automatic sensors. 

```
MQTT_Topic: homeassistant/sbfspot_stp_5000TL_20/sbfspot__xxxxxxxx_
```

Should be. 
```
MQTT_Topic: homeassistant/sbfspot_{plantname}/sbfspot_{serial}
```
It will take Plantname from the options `Plantname: stp_5000TL_20` and get the serial from the inverter.
They are actual variables that the addon will fill in.
************

This may be a buggy typo on my part. The Bluetooth and Ethernet configs are duplicates, but they are separate scripts.
>Invalid value for 'Locale' de-DE|en-US|fr-FR|nl-NL|it-IT|es-ES

It looks like I omitted a default in the file

```
CONFIG_LOCALE="$(bashio::config 'Locale')"
```

```
CONFIG_LOCALE="$(bashio::config 'Locale' 'en-US')"
```

2022.10.1 should have that fixed.

habuild, 2022-10-09:
Hmm the nulls are abit stranger though.

Any chance you could post the setConfig.cfg and SBFspot.cfg that get generated(without sensitive info)
As your first post was using the setConfig.cfg and looked like it was picking up the mqtt data. 

To explain..
setConfig.cfg is the config file SBFspot uses to grab the serial number from the inverter. 
That is created initially and just grabs those couple of sensors. genSenEth sets up that file.

SBFspot.cfg is made by genEthernetConfig.sh so more areas for typos..

karmacop, 2022-10-09:
Found it. MQTT_ItemFormat is also coming across as null.
I put:
MQTT_ItemFormat: "\"{key}\": {value}"
into my config and it's working now.

So once you get that typo fixed I'd say you have ethernet working :)
Thanks.

habuild, 2022-10-09:
Nice, Thanks for that. Should be updated now. 

Do you intend to use addon longterm or were you just trying it out?

karmacop, 2022-10-09:
Long term. I've just set up Home assistant for my parents and this is the only way they can get data from the inverter. Too far away to use bluetooth, modbus doesn't work on this inverter, and there is no webserver on the inverter. I'm happy to help where possible.

habuild, 2022-10-09:
Well good to know this is a solution for them.
Happy to know the ethernet version is actually working, I set it up awhile ago.. but never had a way to test it.

closed 2022-10-09 by karmacop