habuild/hassio-addons #42: Several Blutooth inverter support

lebeck69, 2022-10-29:
serveral ip inverter is apparently already supported but not several BT ones.

HasQT has already started proposed to duplicate SBFspot. I need to test fi mariadb support dual input and PVoutput.
https://community.home-assistant.io/t/sbfspot-bluetooth-homeassistant-addon/402671/184
already thx

habuild, 2022-10-29:
Hasqt here..

I believe mariadb should combine the inverters. Then sbfspot upload will upload a total to pvoutput.

Did you get HAOS-SBFspot and MultiINV-SBFspot to play together?

Sven-Scheers, 2022-11-12:
Hi Hasqt,
Both HAOS-SBFspot and MultiINV-SBFspot are running well on my Hassio. Data are fetched from the inverters, stored in MariaDB and send to PVOutput. PVOutput combine all the data as you said. Great work !!!
Just one question, how can I get data from a third BT Inverter ? :) Can I duplicate one addon easily ?
Thanks

habuild, 2022-11-13:
@Sven-Scheers

Very happy to hear it's working as expected. 

I knew that would open a can of worms for 3 inverters. 

You are obviously welcome to fork and add a second multiinv version to a fork. I would suggest offsetting the time a further 5-10s in the third version. 

So it would be.

HAOS-SBFspot
MultiINV-SBFspot
MultiINV-SBFspot2

Considering.. It's just a copy of multiinv with some renaming. I can properly add it easier than you forking it..

So

I will try to add in the next few days hopefully.

Can you confirm the mqtt sensors are working as expected with two versions running. It should be separate sensors for each inverter.

Sven-Scheers, 2022-11-13:
@habuild 

It would be great if you could make a fork for MultiINV-SBFspot2. 

I confirm you that the mqtt sensors work as expected. There is a specific sensor for each inverter and all entities are duplicated. The mqtt topics are duplicated too. I put some screenshots if it helps you.

[image]
[image]
[image]

Indeed, afterwards, it would be nice to put everything in one addon but I understand that it takes time. 

I could try to help you a bit but for the moment I don't have any free time (new job). In a few weeks, I'll probably have more time for that.

In the meantime, I'm following closely the evolution of the project. I'll come back to you as soon as I have some free time.

Thanks for your work

habuild, 2022-11-13:
Thanks for the screenshots. Very helpful.
Not quite how I expected to be honest. Slightly different from my fudged tests with 1 inverter.

Would you mind posting an entity_id from each addon in. just one will do.

AC power would be reasonable

```
sensor.sma_ac_power
SMA AC POWER
```

I assume you have
for the first inverter
```
sensor.sma_ac_power
SMA AC POWER
```
and for the second inverter

```
sensor.sma_ac_power_1
SMA AC POWER
```

Does that work ok in dev tools > states when you filter the states?
No errors in the logs complaining of non unique IDs or similar?

Yes the quickest/easiest would just be to add a third addon. Its not a fork as such(you can't self fork).

habuild, 2022-11-13:
@Sven-Scheers 

I added TriINV for third inverters. It's essentially a direct copy of MultiINV.
Let me know how you go.
entity name example from all 3 setups would be useful. (see previous message)

For those reading along.

HAOS-SBFspot needs the full default options including PVoutput API and SID.
HAOS-SBFspot is the only addon with SBFspotUpload and hence PVoutput enabled.

MultiINV-SBFspot needs BTaddress and Plantname adjusted for the second inverter.
MariaDB and PVoutput can use the same options as the first inverter to get the options page to save.
MQTT should be the same details as first inverter.
MultiINV-SBFspot has SBFspotUpload and hence PVoutput Disabled. You still need to put an entry in the options to save.

TriINV-SBFspot needs BTaddress and Plantname adjusted for the THIRD inverter.
MariaDB and PVoutput can use the same options as the first inverter to get the options page to save.
MQTT should be the same details as first inverter.
TriINV-SBFspot has SBFspotUpload and hence PVoutput Disabled. You still need to put an entry in the options to save.

habuild, 2022-11-13:
https://github.com/SBFspot/SBFspot/discussions/588

https://community.home-assistant.io/t/sbfspot-bluetooth-homeassistant-addon/402671/232?u=hasqt

@Sven-Scheers

Perhaps multiple addons is actually the wrong way to go about this.

Have you tried with MIS enabled in HAOS-SBFspot and checked if it collects data from all inverters?

lebeck69, 2022-11-13:
I ve tryed MIS option and it didn't worked for me. 
I have SMA inverter 1 netID=04 & inverter 2 SMA netID=04
Only retrieving data of first inverter.
Proposal was coming from here : https://github.com/SBFspot/SBFspot/discussions/588

It's now working with MultiINV-SBFspot
I'm now trying to use MQQT to update energy dashboard instead of PVOUTput.

Sven-Scheers, 2022-11-14:
@habuild 

Here are some screens of the 2 entities. Everything seems to work fine, no error message in the logs.

The MIS option seems interesting because all my inverters use the NetID 04 and it would be easier to group everything together. But I don't know how to enable MIS. MIS_enabled is not in the settings and I don't know how to access SBFspot.cfg.

I'll search a bit and if I don't find it, I'll try TriINV-SBFspot and let you know.

Thanks

habuild, 2022-11-14:
MIS is in the addon settings.
You have to go down to the bottom of the options to open the full list of options. Or just type it in the yaml.

```
MIS_enabled: 1 
```

habuild, 2022-11-15:
@Sven-Scheers

Yes MIS should only need one addon running.
HAOS-SBFspot would be the correct one to use with PVoutput. (I will delete/disable TriINV sometime soon as it shouldn't be needed.) 

>Once the sensors were created manually, I reactivated a single module with MIS activated. The mqtt messages update all the sensors now.

So HAOS-SBFspot is sending out MQTT for all three inverters? Can you post the message from mqtt-Explorer (as text not image, so wrap in back ticks 
``` 
 text

```
To output like this)

```
Text
```

I will have a look at making HAOS-SBFspot pick up the extra sensors for auto sensors (it wasn't implemented prior as no one had been using multiple inverters as fas I know)

Re pvoutput only uploading 1 inverter. I will have to poke around.
Which file did you use to create the DB? Perhaps mine is out of date.
I'm not aware of a setting change that would be needed to change the behaviour.. That doesn't mean it doesn't exit.

Sven-Scheers, 2022-11-15:
@habuild 

Yes HAOS-SBFspot sends MQTT for all inverters. Here are sample messages: 

```
{"PrgVersion": "3.9.7","Plantname": "solarpanel","Timestamp": "2022-11-15T16:30:20+0100","SunRise": "2022-11-15T07:59:00+0100","SunSet": "2022-11-15T16:55:00+0100","InvSerial": 2120149679,"InvName": "SN: 2120149679","InvTime": "2022-11-15T16:30:12+0100","InvStatus": "Ok","InvSwVer": "02.30.06.R","InvClass": "Solar Inverters","InvType": "SB 2500HF-30","InvTemperature": 22.340,"InvGridRelay": "Open","EToday": 1.180,"ETotal": 25256.395,"GridFreq": 50.020,"PACTot": 0.000,"PAC1": 0.000,"UAC1": 237.550,"IAC1": 0.000,"OperTm": 44251.962,"FeedTm": 41608.828,"PDCTot": 0.000,"UDC1": 292.630,"UDC2": 0.000,"IDC1": 0.000,"IDC2": 0.000,"PDC1": 0.000,"PDC2": 0.000,"BTSignal": 65.882,"InvWakeupTm": "2022-11-15T08:01:00+0100","InvSleepTm": "2022-11-15T16:30:19+0100"}

{"PrgVersion": "3.9.7","Plantname": "solarpanel","Timestamp": "2022-11-15T16:35:17+0100","SunRise": "2022-11-15T07:59:00+0100","SunSet": "2022-11-15T16:55:00+0100","InvSerial": 2120088578,"InvName": "SN: 2120088578","InvTime": "2022-11-15T16:35:09+0100","InvStatus": "Ok","InvSwVer": "02.10.18.R","InvClass": "Solar Inverters","InvType": "SB 2500HF-30","InvTemperature": 0.000,"InvGridRelay": "Information not available","EToday": 1.702,"ETotal": 28976.374,"GridFreq": 0.000,"PACTot": 0.000,"PAC1": 0.000,"UAC1": 0.000,"IAC1": 0.000,"OperTm": 44473.210,"FeedTm": 42022.200,"PDCTot": 0.000,"UDC1": 0.000,"UDC2": 0.000,"IDC1": 0.000,"IDC2": 0.000,"PDC1": 0.000,"PDC2": 0.000,"BTSignal": 0.000,"InvWakeupTm": "2022-11-15T16:32:05+0100","InvSleepTm": "2022-11-15T16:34:13+0100"}

{"PrgVersion": "3.9.7","Plantname": "solarpanel","Timestamp": "2022-11-15T16:35:17+0100","SunRise": "2022-11-15T07:59:00+0100","SunSet": "2022-11-15T16:55:00+0100","InvSerial": 2120070967,"InvName": "SN: 2120070967","InvTime": "2022-11-15T16:35:12+0100","InvStatus": "Ok","InvSwVer": "02.10.18.R","InvClass": "Solar Inverters","InvType": "SB 2500HF-30","InvTemperature": 0.000,"InvGridRelay": "Information not available","EToday": 1.771,"ETotal": 26632.621,"GridFreq": 0.000,"PACTot": 0.000,"PAC1": 0.000,"UAC1": 0.000,"IAC1": 0.000,"OperTm": 44243.839,"FeedTm": 41377.609,"PDCTot": 0.000,"UDC1": 0.000,"UDC2": 0.000,"IDC1": 0.000,"IDC2": 0.000,"PDC1": 0.000,"PDC2": 0.000,"BTSignal": 0.000,"InvWakeupTm": "2022-11-15T16:31:36+0100","InvSleepTm": "2022-11-15T16:34:46+0100"}
```

I had used this file to create the db: https://github.com/SBFspot/SBFspot/blob/master/SBFspot/CreateMySQLDB.sql

I will try to change the configuration for PVoutput, I'll let you know.

It already works at 90%, it's great. Thanks for your support.

habuild, 2022-11-16:
@Sven-Scheers 

It was late when I first read this on mobile.
I see what you mean by manually created the sensors now (manually created to me means by hand in yaml, the old way)
You ran each inverter through the auto sensor creation.

So when you say
>By running the addon successively without MIS and with the parameters of the first, the second and the third inverter,

Do you mean you just changed in the HAOS-SBFspot options, the BTaddress and Plantname or did you change the MQTT_topic as well?

for example for inverter2
```
BTaddress: 89:89:00:00:89:89            ## The BT address of the second inverter.
Plantname: solarplant                       ## You left this the same for all 3 inverters?
MQTT_topic: sbfspot_solarpanel       ## or was this sbfspot_solarpanel2?
```

Perhaps I will just write that up as a documentation change and not try to make it make all 3 at once. It is relatively rare that one should be making the sensors and even less common for multi inverters.

Alternatively I may add to here

[image]

Create primary 
create secondary 
create tertiary

That would probably work best if I add 
to the options 
some extra space for the second and third inverter ~names

MaxMSchneider, 2022-12-22:
@Sven-Scheers @habuild 

I am also running in an issue during the sensor creation. Weather I use HAOS-SBFspot or MultiINV-SBFspot, it does not change the behaviour. Here is my problem:

I have 3 SB 4000TL-20, all close to each other, all sitting on NetID=5 and I want to connect via BT. Running HAOS on an RPi 3+.

When I run the addon to create the MQTT Sensors, the first one runs just fine. Then I change the configuration via the addon GUI and fill in the BT-Mac as well as the plant name. I tried even leaving the plant name the same as well, but whatever I do, as soon as I run the addon again to create the sensors for my second inverter, the log prints the following:

```
[13:39:46] INFO: Using Existing Device info
{
  "InvSerial": 2100230XXX,
  "InvName": "SN: 2100230XXX",
  "InvClass": "Solar Inverters",
  "InvType": "SB 4000TL-20",
  "InvSwVer": "03.01.03.R"
}
```
So what happens is, that not a new device senso is created, it just keeps publishing into the old mqtt sensor topic of the first inverter (also using the SN of the first inverter). This is weird.

So far I've been trying pretty much everything you could think of. Even to hard coding the sensors into the configuration.yaml as subscriptions to the mqtt topics.

That kinda worked, but once I ran HAOS-SBFspot with MIS=1 I just received the data from 2/3 inverters.

@Sven-Scheers What was your procedure to get all 3 inverters in your MQTT addon like in your screenshots?

habuild, 2022-12-22:
@MaxMSchneider

There is a work around of sorts for that existing info. It is a file stored in `/data/device`

If you ssh into the addon you can remove that device file. (you need to use the community ssh addon with protection mode off, not the official one) 

```
docker exec -it addon_a51a23d8_haos-sbfspot /bin/bash

rm /data/device
```

You can use mqtt-Explorer to remove the retained message under haos-sbfspot with the same info. 

Then you can create the sensors. You would need to repeat this process for the third inverter as the file would be created again for the second inverter.

MaxMSchneider, 2022-12-23:
@habuild 

Thank you so much for your quick response! All worked out and is working as intended. You kinda saved Christmas. absolute hero!

Thanks a lot and Happy Holidays!

jhilgenberg, 2023-06-26:
Your add-on is also in 2023 in use, it works perfectly! But also for me there is an issue with multiple bluetooth inverters. The MIS_Enabled option isn't working as expected. Is there any possibility to receive a clone of the MultiNV-SBFspot add-on so i can use my third inverter?

Thanks in advance!

habuild, 2023-06-27:
@jhilgenberg

Hi Julian,

What's not working as expected? If you can add some details about the problem there may be a simple config issue to fix.

jhilgenberg, 2023-06-27:
> @jhilgenberg
> 
> Hi Julian,
> 
> What's not working as expected? If you can add some details about the problem there may be a simple config issue to fix.

Thanks for your response. I‘ve first added all inverters one by one with the „create“ option. After that I switched this to „No“ and set the MIS_Enabled to 1. Sadly only the inverter set in the configuration gets updated not the other two..

habuild, 2023-06-27:
@jhilgenberg

Have you had MIS working when using SBFspot previously in a different setup? If you have a spare sd card or pc etc, it might be worth getting MIS working in the proper SBFspot install to test MIS.

Have you set the Net_ID of your inverters so that they make the Bluetooth network?

The values that show up in your current setup, is it the values of only one inverter or is it all of them combined together?

jhilgenberg, 2023-06-27:
The inverters we're in default configuration. I first connected with the sunny explorer and it asked me to change the password, so i think the NET_ID is not set on the inverters. Can i do it with the Sunny Explorer software?

habuild, 2023-06-27:
I believe so.. I'm not actually sure. I don't have multiple inverters. I was trying to google it for you without much success atm.

xtrapowa, 2024-01-28:
@Sven-Scheers 

Hi, I have a similar configuration but with 2 inverters. I have the same problem with SBFspot, only information from 1 inverter is send to pvoutput.
[image]

My config for pvoutput is
pvoSID: "2120199816:mySID,2120199858:mySID"

Have you now a solution for that ?

Thank you !

SleXik

habuild, 2024-01-28:
@xtrapowa

```
Inverter1:acc1,Inverter2:acc2
```

I believe you are required to have multiple pvoutput accounts, not a single account. 

https://pvoutput.org/help/data_aggregation.html

https://forum.pvoutput.org/t/adding-a-second-inverter-more-panels/5166

https://forum.pvoutput.org/t/2-systems-combine-total-generation-is-it-possible/780

xtrapowa, 2024-01-28:
@habuild ,

Hi, ok I'll try this way :) and let you know.

Thank you

Merwases, 2025-09-24:
How to get 2 separate mqtt instances, one for each Inverter ?
All I tried ends up showing both inverter data at the log, but not at the mptt machines (Geräte).
I try to show some 2 screens about it.
May be tomorrow when inverters will awake, it shows data 🤷🏼‍♂️.

[image]
[image]

Merwases, 2025-09-25:
Anything I try, no way.
Logs showing entities/data of both inverter, but always  pointing to one, despite I renamed it at creation time.
That's a typic mqtt connection log:

MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}),"PrgVersion": "3.9.12","Plantname": "**SMA3000**","Timestamp": "2025-09-25T15:45:12+0200","SunRise": "2025-09-25T07:18:00+0200","SunSet": "2025-09-25T19:16:00+0200","InvSerial": 2120096870,"InvName": "SN: **2120096870**","InvTime": "2025-09-25T15:32:40+0200","InvStatus": "Ok","InvSwVer": "02.30.06.R","InvClass": "Solar Inverters","InvType": "SB 3000HF-30","InvTemperature": 32.290,"InvGridRelay": "Closed","EToday": 0.751,"ETotal": 51250.588,"GridFreq": 49.970,"PACTot": 127.000,"PAC1": 128.000,"UAC1": 236.300,"IAC1": 0.542,"OperTm": 58154.085,"FeedTm": 55057.736,"PDCTot": 137.000,"UDC1": 425.450,"UDC2": 0.000,"IDC1": 0.322,"IDC2": 0.000,"PDC1": 137.000,"PDC2": 0.000,"BTSignal": 66.667,"InvWakeupTm": "2025-09-25T07:36:25+0200","InvSleepTm": "2025-09-25T15:32:46+0200"
Client null sending CONNECT
Client null received CONNACK (0)
Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_SMA3000/sbfspot_2120096870', ... (768 bytes))
Client null sending DISCONNECT
MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}),"PrgVersion": "3.9.12","Plantname": "**SMA3000**","Timestamp": "2025-09-25T15:45:12+0200","SunRise": "2025-09-25T07:18:00+0200","SunSet": "2025-09-25T19:16:00+0200","InvSerial": 2120108811,"InvName": "SN: **2120108811**","InvTime": "2025-09-25T15:40:53+0200","InvStatus": "Ok","InvSwVer": "02.30.06.R","InvClass": "Solar Inverters","InvType": "SB 2500HF-30","InvTemperature": 31.960,"InvGridRelay": "Closed","EToday": 0.511,"ETotal": 35372.121,"GridFreq": 49.970,"PACTot": 87.000,"PAC1": 86.000,"UAC1": 235.530,"IAC1": 0.369,"OperTm": 57517.968,"FeedTm": 53929.323,"PDCTot": 107.000,"UDC1": 314.720,"UDC2": 0.000,"IDC1": 0.340,"IDC2": 0.000,"PDC1": 107.000,"PDC2": 0.000,"BTSignal": 0.000,"InvWakeupTm": "2025-09-25T07:44:42+0200","InvSleepTm": "2025-09-25T15:40:59+0200"
Client null sending CONNECT
Client null received CONNACK (0)
Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_SMA3000/sbfspot_2120108811', ... (765 bytes))
Client null sending DISCONNECT
********************
* ArchiveDayData() *
********************
startTime: 25/09/2025 00:00:00
Thu Sep 25 15:45:17 2025: INFO: Done.
[15:45:45] INFO: [SBFspot Upload Log Latest]

Merwases, 2025-09-25:
**_I type my answers into your text.
Thanks a lot for guiding me despite being a little noobish._**

> You removed the existing device before you tried adding the individual devices? 

**_Yes, at the mqtt addons "devices"._**

in MQTT explorer you will probably need to delete HAOS-SBFspot device topic. 

**_Hmmm... I did this once, but if your addon** isn't running, no "topic" at mqtt explorer, isn't it ?_**

That is only used to configure the sensors. **The actual sensors end up in the homeassistant/sensor/sbfspot_yourplantname/sbfspot_yourSerial_eachSensor/config**
> 
**_Okay... wherever this is located, it's not to find for me..._**

> Try this. Stop the addon. Set sensor creation to NO. Set the config/settings to your first inverter.
> 
> Delete anything related to Your inverter in MQTT integration in home assistant, You can also delete this from MQTT explorer.
> 
_**I did this felt 100 times...**_

> Publish these two topics and messages. These need to match what the addon has been sending to MQTT exactly, any typos will make this not work and the sensors will be unknown. These messages should be set to RETAIN.

**_How to "publish" and at which place ? (sory, but I'm not a programmer 🫣)_**

> 
> Inverter 1
> 
> ```
> TOPIC
> homeassistant/sensor/sbfspot_3000HF/sbfspot_2120096870EToday/config
> 
> MESSAGE.
> 
> {"name": "SMA Energy Daily", "state_topic": "homeassistant/sbfspot_!!!!YOURPLANTNAME!!!!!/sbfspot_212009687", "value_template": "{{ value_json.EToday | float(default) }}", "unique_id": "2120096870_EToday", "device_class": "energy", "state_class": "total_increasing", "unit_of_measurement": "kWh", "enabled_by_default": "true", "icon": "mdi:solar-power-variant", "device": { "identifiers": ["HAOS-SBFspot-Sensors"], "name": "HAOS-SBFspot", "model": "SB 3000HF-30", "manufacturer": "SMA", "sw_version": "03.10.37.R" }}
> ```
> 
> ```
> /sbfspot_!!!YOURPLANTNAME!!!/ 
> needs to be edited to what you are actually calling the plant names.
> 
> eg
> 
> /sbfspot_BobsInverter1/
> 
> This needs to match what the addon currently send to mqtt explorer.
> ```
> 
> Inverter 2
> 
> ```
> TOPIC
> homeassistant/sensor/sbfspot_5000SBhere/sbfspot_2120108811EToday/config
> 
> MESSAGE.
> 
> {"name": "SMA Energy Daily", "state_topic": "homeassistant/sbfspot_!!!!YOURPLANTNAME!!!!!/sbfspot_2120108811", "value_template": "{{ value_json.EToday | float(default) }}", "unique_id": "2120108811_EToday", "device_class": "energy", "state_class": "total_increasing", "unit_of_measurement": "kWh", "enabled_by_default": "true", "icon": "mdi:solar-power-variant", "device": { "identifiers": ["HAOS-SBFspot-Sensors"], "name": "HAOS-SBFspot", "model": "SB 5000 or whatever", "manufacturer": "SMA", "sw_version": "03.10.37.R" }}
> 
> 
> adjust the plantname to match your second inverter plantname
> ```
> 
> Now start the addon and see if the messages get data.
> 
> These messages should create two devices in your HA MQTT integration.

Merwases, 2025-09-25:
Thanks.
My head is smoking already about failed tyings 🤣.

I try this at a later time using my laptop as working at the phone till now is not best at copy/paste and always closing HA App if switching to other stuff.

What I'm wondering about it, that everything is shown at the log = has been created.
But despite any hard tries, alsways just to one device.
(Shown at the screens)
This is shown at my last pasted log too...

habuild, 2025-09-25:
> Both devices are named SMA3000, despite the 2. was named SMA2500 during setup/sensor creation.
That's the point it's not showing 2 divices at mqtt I belive mostly.

Yes that is looking like the problem.

In your earlier post I also see. They are the messages being sent by the addon to the MQTT server. If you adjust those earlier messages I said to publish. You get energy today values. The HAOS_addon will poll/query the inverter when it first restarts. So just restart the addon if its night time when you play around with making the messages.

```
Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_SMA3000/sbfspot_2120096870', ... (768 bytes))

Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_SMA3000/sbfspot_2120108811', ... (765 bytes))

```

If you make your topics.

Inverter 1
```
TOPIC
homeassistant/sensor/sbfspot_SMA3000/sbfspot_2120096870EToday/config

MESSAGE.

{"name": "SMA Energy Daily", "state_topic": "homeassistant/sbfspot_SMA3000/sbfspot_212009687", "value_template": "{{ value_json.EToday | float(default) }}", "unique_id": "2120096870_EToday", "device_class": "energy", "state_class": "total_increasing", "unit_of_measurement": "kWh", "enabled_by_default": "true", "icon": "mdi:solar-power-variant", "device": { "identifiers": ["HAOS-SBFspot-Sensors"], "name": "HAOS-SBFspot", "model": "SB 3000HF-30", "manufacturer": "SMA", "sw_version": "03.10.37.R" }}

```

and 
Inverter 2

```
TOPIC
homeassistant/sensor/sbfspot_SMA3000/sbfspot_2120108811EToday/config

MESSAGE.

{"name": "SMA Energy Daily", "state_topic": "homeassistant/sbfspot_SMA3000/sbfspot_2120108811", "value_template": "{{ value_json.EToday | float(default) }}", "unique_id": "2120108811_EToday", "device_class": "energy", "state_class": "total_increasing", "unit_of_measurement": "kWh", "enabled_by_default": "true", "icon": "mdi:solar-power-variant", "device": { "identifiers": ["HAOS-SBFspot-Sensors"], "name": "HAOS-SBFspot", "model": "SB 5000 or whatever", "manufacturer": "SMA", "sw_version": "03.10.37.R" }}

```

That should be matching what the addon is posting. Once one sensor is working, You just have to publish the other sensors with slight adjustments. So nearly there.

Merwases, 2025-09-25:
If the config could be edited to SMA2500 for the 2. inverter, may be doing the trick ?
But I even don't find the config by file editor despite showing root and so on... 😢.

habuild, 2025-09-25:
Try this and let me know if you get data when the addon restarts

Inverter 1

```
TOPIC
homeassistant/sensor/sbfspot_SMA3000/sbfspot_2120096870EToday/config

MESSAGE.

{"name": "SMA Energy Daily", "state_topic": "homeassistant/sbfspot_SMA3000/sbfspot_212009687", "value_template": "{{ value_json.EToday | float(default) }}", "unique_id": "2120096870_EToday", "device_class": "energy", "state_class": "total_increasing", "unit_of_measurement": "kWh", "enabled_by_default": "true", "icon": "mdi:solar-power-variant", "device": { "identifiers": ["HAOS-SBFspot-3000"], "name": "HAOS-SBFspot3000", "model": "SB 3000HF-30", "manufacturer": "SMA", "sw_version": "03.10.37.R" }}

```
and
Inverter 2

```
TOPIC
homeassistant/sensor/sbfspot_SMA2500/sbfspot_2120108811EToday/config

MESSAGE.

{"name": "SMA Energy Daily", "state_topic": "homeassistant/sbfspot_SMA3000/sbfspot_2120108811", "value_template": "{{ value_json.EToday | float(default) }}", "unique_id": "2120108811_EToday", "device_class": "energy", "state_class": "total_increasing", "unit_of_measurement": "kWh", "enabled_by_default": "true", "icon": "mdi:solar-power-variant", "device": { "identifiers": ["HAOS-SBFspot-2500"], "name": "HAOS-SBFspot2500", "model": "SB 5000 or whatever", "manufacturer": "SMA", "sw_version": "03.10.37.R" }}
```

Merwases, 2025-09-25:
I deleted all stuff and setup just SMA3000 at the moment, to get a clean setup.
Still not really understand too how to use that topic stuff (at mqtt explorer ?... an which steps by step before...)

habuild, 2025-09-26:
@Merwases 

Hey So I have something for you to try. Its on my test account so you will need to add the repository in your addon shop.

 [habuild/hassio respository](https://github.com/habuild/hassio)

[hassio test addon](https://github.com/habuild/hassio/tree/main/haos-sbfspot)

It's a future update that I have yet to push to the main repository(which is habuild/hassio-addons to avoid future confusion)

You shouldn't need to set up much at all beyond what you already have. 

To setup you second set of sensors, 
Add your second inverter settings into the options and leave MIS_enabled at 0
You need to run the addon once in Create mode. 
Then once in Multi mode.

Then set the addon to NO mode and enable MIS and let me know if you get data going.

habuild, 2025-09-30:
@Merwases 

[Stand alone MQTT Discovery Msg Creation](https://github.dev/habuild/hassio/tree/main/sensorsgen)

You can try this out too. Also in my test repo for the time being.

scottpancakes2, 2025-10-24:
G'day
Firstly thanks for putting this addon together! its been super helpful for me in the past with multiNV-spot, until that was depreciated.

I've been trying to get HAOS-SBFspot & HAOS-SBFspot SensorsGen to recognise my two inverters using the method you've described about a month ago

> To setup you second set of sensors,
Add your second inverter settings into the options and leave MIS_enabled at 0
You need to run the addon once in Create mode.
Then once in Multi mode.
Then set the addon to NO mode and enable MIS and let me know if you get data going.

I've tried to use MQTT to published a discovery packet  to create the sensors, both with no success.

Can someone provide a more 'idiots guide' into getting this working with 2 inverters as I'm really struggling to get HAOS-SBFspot to recognise more than one inverter. 

I owe anyone who can help a beer!

habuild, 2025-10-24:
@scottpancakes2

Sorry not sure if you deleted your comment or what? It's showing up in my email but not here. 

Anyway

I guess I shall need to clean up the instructions on the add-on eventually. 

So with sensorsgen all that does it make the second and third inverter sensors for mqtt. You can run it once and stop/uninstall it when you have the sensors you need. 

The messages and test message are just debugging tools. You can ignore them for your purposes. 

Firstly setup your first inverter in haos-sbfspot with the sensors. You will probably need to do this with MIS_enabled turned off so it doesn't accidentally extra inverter 2 sensors (not something I can test as I only have 1 inverter) . Anyway get you first inverter sensors setup. 

Then go to the sensorsgen add-on 
You can enable MIS now, so that you are getting both MQTT sent from haos-sbfspot.

You must put your first inverter details in the settings precisely.

Put in your info for your second inverter. 
Serial number is the important one. The rest are sort of cosmetic for how the MQTT device displays. 
Plant name will auto-generate if you leave it as null.

Then enable the make sensors and it should create the second device in the MQTT integration in home assistant.

You should be good after that assuming the topics got created correctly. That's what the test messages can be used for. You shouldn't need the test messages though.

Merwases, 2025-10-24:
If I remember right, 1. inverter has been set up with MIS 0 and after 2. inverter setup, I set MIS to 1.