habuild/hassio-addons #47: Error starting sbfspot addon

jks1981, 2022-11-29:
I keep getting an error starting sbfspot:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sbfspot (no readiness notification)
s6-rc: info: service legacy-services successfully started
[07:34:45] INFO: 
[07:34:45] INFO:  Linux 5.15.61-v7l #1 SMP Fri Oct 21 00:14:01 UTC 2022 armv7l
[07:34:45] INFO: 
|   INITIALISE MQTT SENSOR CREATION   |
sma
core-mosquitto
mqtt
homeassistant/sbfspot_{plantname}/sbfspot_{serial}
PrgVersion,Plantname,Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvSwVer,InvClass,InvType,InvTemperature,InvGridRelay,EToday,ETotal,GridFreq,PACTot,PAC1,UAC1,IAC1,OperTm,FeedTm,PDCTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2,BTSignal,InvWakeupTm,InvSleepTm
[07:34:46] INFO: Using Existing Device info
parse error: Invalid numeric literal at line 1, column 47
[06:34:46] WARNING: Halt add-on with exit code 4
s6-rc: info: service legacy-services: stopping
[06:34:46] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

My configuration:

Connection_Type: Ethernet
Sensors_HA: Create
BTAddress: fake
Password: ********
LocalBTAddress: fake
IP_Address: 192.168.178.133
Plantname: sma
Latitude: "49.9"
Longitude: "10.2"
Timezone: Europe/Brussels
DateTimeFormat: "'%H:%M:%S %d-%m-%y'"
SQL_Password: fake
MQTT_User: ********
MQTT_Pass: ********
MQTT_Topic: homeassistant/sbfspot_{plantname}/sbfspot_{serial}
MQTT_Data: >-
  PrgVersion,Plantname,Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvSwVer,InvClass,InvType,InvTemperature,InvGridRelay,EToday,ETotal,GridFreq,PACTot,PAC1,UAC1,IAC1,OperTm,FeedTm,PDCTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2,BTSignal,InvWakeupTm,InvSleepTm
PVoutput_SID: fake
PVoutput_Key: fake
LogLevel: info

habuild, 2022-11-29:
It connects and prints to the log when not trying to create sensors, is that correct?

Does it print the serial number to the log?

The sensor creation uses the serial number to create the unique ID for the sensors. So if that isn't showing up I may have to fix something..

Try a purge then retry create sensors. The device file gets created in /data/device purge should delete that so it re creates that file. If that fails try an uninstall / reinstall. That will delete anything in data.

jks1981, 2022-11-29:
ok, a reinstall did the trick.

another question: is it possible to use sqlitedb instead of mariadb?

habuild, 2022-11-29:
>ok, a reinstall did the trick.

Nice.

>another question: is it possible to use sqlitedb instead of mariadb?

No, that requires recompiling of SBFspot in the addon via the docker file. 

>is the sbfspot.cfg created at every (re-)start?

Yes, it's sort of a limitation of how to get the options from options.json(yaml) which is then converted to JSON, and then converted again into the SBFspot.cfg. It's also mostly my lack of ability probably...

This addon is effectively the same as  https://github.com/SBFspot/sbfspot-config it just creates the config files. The SBFspot binary is unchanged from the the official version.

closed 2022-11-29 by habuild