habuild/hassio-addons #50: Export CSV Data to Homeassisstant www folder

virusmoere, 2022-12-23:
**Is your feature request related to a problem? Please describe.**
I am using the CSV export feature of SBFspot for a legacy app called "sunwatcher". Unfortunately the CSV files are only accessible within the container. The app requires access to the files via http/https or ftp. Home Assistant has a public folder available in the config directory via HTTP integration (https://www.home-assistant.io/integrations/http/#hosting-files)
**Describe the solution you'd like**
Map the config volume to the container via the config.yaml

config.yaml:
```
map:
- config:rw
```

habuild, 2022-12-24:
I don't think this is a good idea security wise. This exposes all of your config directory including secrets to the addon, which is on the host network.

/data is already available on the host.

You can ssh into the addon or host to access this directory.
```
 mnt/data/supervisor/addons/data/a51a23d8_haos-sbfspot
```

closed 2022-12-24 by habuild

virusmoere, 2022-12-24:
Hi,

I understand. But I am unable to access this data without creating unsupported docker containers or using some hacky way to access the data on home assistant os.

Would it be a solution to use other predefined mappings according to the documentation like /addons or /share? Those are also available to the home assistant container and I could copy the needed data to the www directory from there without compromising security.

habuild, 2022-12-26:
Yeah i will add share for the next update.