habuild/hassio-addons #68: HAOS sbfspot stopes after connection

FritsOV, 2023-06-07:
i have installed for some time now the haos-sbfspot on an pi4. 
I noticed a problem with connecting with the inverter. Especially with the latest HAS update. I get an error 250 or 101 and 107. 
I reinstalled spot but this did not change anything. 
I tested with solaroakly on my android and got connection with no problem and all data is comming in fine. I tested with the sunnyexplorer and also no problem.
What is error 250, 101 and 107?

System halts with an error. 
[10:46:26] INFO: 
[10:46:26] INFO: 
[10:46:26] INFO:  ||    Generating HA Sensors    ||
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v -finq -mqtt -cfg/usr/bin/sbfspot/SetConfig.cfg
Reading config '/usr/bin/sbfspot/SetConfig.cfg'
Wed Jun  7 10:46:26 2023: INFO: Starting...
sunrise: 05:18
sunset : 21:50
Connecting to 00:80:25:18:51:0C (1/10)
Initialising...
SUSyID: 125 - SessionID: 930237279
SMA netID=02
Serial Nr: 4194C084 (1100267652)
BT Signal=72.2%
Logon OK
SUSyID: 99 - SN: 1100267652
Device Name:      SN: 1100267652
Device Class:     Solar Inverters
Device Type:      SB 3300TL HC
Software Version: 12.12.203.R
Packet status: 21
SUSyID: 99 - SN: 1100267652
Device Status:      Ok
Packet status: 21
SUSyID: 99 - SN: 1100267652
Device Temperature: Information not available
Packet status: 21
getGridRelayStatus returned an error: 21
recv() returned an error: 110
getOperationTime returned an error: -1
send() returned an error: 107

*

FritsOV, 2023-06-07:
solved after restoring a backup
Still like to know where the errorcodes stand for e.g. 250, 101 and 107

habuild, 2023-06-07:
This is what I find when I google the errors

Error 110 = Connection timed out
Error 107 = Transport endpoint is not connected

Error 250 (unsigned) = -6 (signed) which is "E_INIT or Initialization Error" SBFspot was not able to start the communication with the inverter. This error should be ignored and connecttion should be retried later.

I didn't really have any specific suggestions earlier so glad you got it sorted.

It might possibly have a conflict with the database or something along those lines.

closed 2023-06-07 by habuild