habuild/hassio-addons #74: Setup for SB 5000TL-20 on Raspberry Pi problems

3gfisch, 2023-07-15:
Hi, sorry if this is not the right place but i could not find a good place for support or instructions for my problems. Thanks for providing this project already!
As far as i understood this add on is to connect SMA inverter like Sunny Boy SB 5000TL-20 via Bluetooth to HA?
In general (like in the other issue) i don’t need any DB or account where the data is pushed to, if energy/power data is available in HA that good for me, as far as i understood this can be done if i leave the configurations blank or put in fake info right? And there is no other more light weight add on for this use case?

First i tested with the Sunny Explore on PC that the general BT connection to the inverter works. It connected a the BT LED light up on the inverter. But each time it asked me to change the default password (i assume it didnt changed it??) and i could not se munch info or any power or energy data..

My setup:
Raspberry Pi 4 with HA OS freshly installed last week.
No DB or anything setup up, only the add on and some others like file editor or Terminal…

A few notes to the setup and problem i faced (which can maybe be improved):
With the UI i was told that i can leave fields blank but if i did i got errors while saving or starting.
Took me a while to find the BT MAC address of the Pi only to notice later i can put in “” if i use the YAML config, but there i can’t see the hints wich are shown in the UI. Also to find the MAC of the inverter took me a PC an BT scanner (normally i use HA via the App and a iPad), could this not be done via a drop down menu of the available BT devices? Or a Terminal command provided in the instructions?
Maybe im too much a beginner with all this stuff and others don’t struggler so long wich such problems i do then please ignore this..
Any way i think i got the config but please double check, thanks, (xxx changed for this post): 

```
Connection_Type: Bluetooth
Sensors_HA: "No"
BTAddress: 00:80:25:08:x:xx
Password: "0000"
LocalBTAddress: ""
IP_Address: ""
Plantname: sb5000tl-20
Latitude: "49.xx"
Longitude: "9.xx"
Timezone: Europe/Berlin
DateTimeFormat: "'%H:%M:%S %d-%m-%y'"
SQL_Password: mariadb_pwFake
MQTT_User: mqttxxx
MQTT_Pass: mqttxxx
MQTT_Topic: homeassistant/sbfspot_{plantname}/sbfspot_{serial}
MQTT_Data: >-
  PrgVersion,Plantname,Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvSwVer,InvClass,InvType,InvTemperature,InvGridRelay,EToday,ETotal,GridFreq,PACTot,PAC1,UAC1,IAC1,OperTm,FeedTm,PDCTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2,BTSignal,InvWakeupTm,InvSleepTm
PVoutput_SID: 2100001xxx:fake
PVoutput_Key: pvoAPIkeyFake
LogLevel: info
``` 
So all with fake is in the YAML like this but there is no PVout ID or key or mariadb setup, hope this is not the problem?
The inverter BT LED is shortly flashing up if i start the add on so i think it’s trying to connect to the correct device.

Log from add on:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sbfspot (no readiness notification)
s6-rc: info: service legacy-services successfully started
[00:07:07] INFO: 
[00:07:07] INFO:  Linux 6.1.21-v8 #1 SMP PREEMPT Thu Jun 15 17:49:15 UTC 2023 aarch64
[00:07:07] INFO: 
|   SKIPPING MQTT SENSOR CREATION   |
[00:07:15] INFO: 
[00:07:15] INFO: 
[00:07:15] INFO:  [Host Bluetooth MAC Address] Controller DC:A6:32:56:xx:xx homeassistant [default]
[00:07:15] INFO: 
[00:07:15] INFO: 
[00:07:15] INFO:  ||    Generating Bluetooth Configs    ||
[00:07:15] INFO:  ||    /usr/bin/sbfspot/SBFspot.cfg    ||
[00:07:15] INFO:  || /usr/bin/sbfspot/SBFspotUpload.cfg ||
[00:07:15] INFO: 
[00:07:15] INFO:  [Starting SBFspotUpload]
[00:07:15] INFO: [Starting cron in foreground]
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v -ad0 -am0 -mqtt -finq
Reading config '/usr/bin/sbfspot/SBFspot.cfg'
Sun Jul 16 00:07:45 2023: INFO: Starting...
sunrise: 05:36
sunset : 21:21
Connecting to 00:80:25:08:xx:xx (1/10)
Initialising...
SUSyID: 125 - SessionID: 844193932
SMA netID=01
Sun Jul 16 00:07:46 2023: CRITICAL: Failed to initialise communication with inverter.
Sun Jul 16 00:07:46 2023: INFO: Done.
Configuration Error: Syntax error on line 34 [/usr/bin/sbfspot/SBFspotUpload.cfg]
``` 

>SKIPPING MQTT SENSOR CREATION
Why this? Is this expected, i thought the data is provided via MQTT to HA? The login info is set like in HA MQTT broker and works with the AhoyDTU (used for a different device)

>CRITICAL: Failed to initialise communication with inverter.
I think this is my main problem, but how can i debug it further? I tried default password and the one i put in multiple times for the Sunny Explore with same result.

>Syntax error on line 34 [/usr/bin/sbfspot/SBFspotUpload.cfg]
After checking like below i think its from the wrong login info PVoutput_SID but hopefully will not crash everything?
https://github.com/habuild/hassio-addons/issues/42#issuecomment-1314311997

Thanks a lot for any help!

habuild, 2023-07-16:
Hi, yes here is an appropriate to ask for support. I will try and run through your list of questions in order.

1. Yes this addon helps to configure SBFspot, and output it's data to HA. This project has no official affiliation with SBFspot, It is just an addon to make the config for SBFspot and run it in a "*docker style container*" in HAOS. I wrote it as user of both HA and SBFspot who wanted to add this data to HA.
  
   [SBFspot Official github](https://github.com/SBFspot/SBFspot) "SBFspot, formerly known as SMAspot, is an open source project to get actual and archive data out of an SMA® inverter over Bluetooth or Ethernet (Speedwire®) It works on either Linux ([Raspberry Pi](http://www.raspberrypi.org/)) and Windows." This is a quote from the Standalone version of SBFspot.
So Yes it is designed to interface SMA inverters. 

2. There are lighter weight versions of SBFspot that don't upload to PVoutput and use different databases, and include MQTT functionality. They are not setup to run as docker containers at this date, and require a different install method. usually [SBFspot config on a PI](https://github.com/SBFspot/sbfspot-config) or Windows installs. You could use the MQTT function to send your data to MQTT broker this way.

3. Yes you can put fake [PVoutput](https://pvoutput.org/) account data so as not to use PVoutput while using this addon.

4. This is essentially how the MQTT part works. Data is sent to the broker from SBFspot and then.. The other side being HA needs to create the config to read this data from the MQTT broker(or server). These can either be done as manual MQTT entities or using MQTT Auto discovery. I was using [ manual entities ](https://github.com/habuild/hassio-addons/blob/main/.images/sbfspot_sensors.yaml) originally. I changed this addon to "create the sensors using the MQTT auto discovery feature of HA. 
   
   This relates to the SKIPPING MQTT SENSOR CREATION question.
   As I said above you can have manual sensors or automatic sensors. 
   The addon has 3 options for sensors, NO, CREATE and PURGE.

   The NO option is for either manual sensors or for AFTER you have created the auto sensors.

   The CREATE option creates the AUTO sensors from your options in your MQTT Broker and then continues to run the addon as Normal, you don't need to continue using this option after the sensors have been created. You could in theory create the sensors and discontinue use of this addon and run SBFspot on a windows machine with matching MQTT config.
    
   The PURGE option deletes the AUTO sensors from the MQTT Broker and subsequently from HA. This would have no effect on manually created sensors. 

5. The host BT device mac address should be showing up in Home Assistant -> Settings -> Devices & Services -> Bluetooth -> Device 
   It also gets listed in the addon log `[Host Bluetooth MAC Address] Controller DC:A6:32:56:xx:xx homeassistant [default]`
   
   The Inverter MAC I usually just get via my phone personally. No real easy way I am aware of to make it easy to find(such as a drop down menu).

Until now I have basically covered the general concepts. Now I will speak more to your specific issues. 

>Syntax error on line 34 [/usr/bin/sbfspot/SBFspotUpload.cfg]

You are correct here. The addon will still post MQTT data even with this error.

>First i tested with the Sunny Explore on PC that the general BT connection to the inverter works. It connected and the BT LED light up on the inverter. But each time it asked me to change the default password (i assume it didnt changed it??) and i could not se munch info or any power or energy data..

Sunny Explorer is the Official SMA(the Inverter Manufacturer) produced program for reading data from your Inverter. This connection appears to not be working from the info you have provided due to an incorrect password. The default user password used to be 0000, however there was an update to sunny explorer in the not so distant past that requires a more complex password to be created. This is also the password you will need for SBFspot. If you cannot connect with Sunny explorer you need to figure it out with the correct password. There should be data in SMA explorer, although SBFspot and this addon, have nothing to do with SMA Sunny explorer or the SMA manufacturer. Perhaps I have misunderstood you here.

>CRITICAL: Failed to initialise communication with inverter.

Here you would normally see password or BT error if it was a password error or incorrect BT MAC. So I would suggest trying to move your rPI closer to the inverter to see if that helps. It might also be worth tryin the windows version of SBFspot to confirm it will connect to your inverter. Also make sure Sunny explorer is not already connected to the inverter.

3gfisch, 2023-07-18:
Thanks for the detailed answers :)
I now tried and put it directly next to the inverter, but still getting the same error message. Is there some way to debug further what might be the problem?

Yes im aware that Sunny Explorer is the official tool. I thought it “works” because at least the BT LED is saying on if i used it and it shows connected and signal is good. But it also shows things which look “bad” so i don’t know how to judge this, maybe this first has to work before going on with SBFspot.
Each time i start the SE it tells me to set up more complex password, but don’t complains that it could not set it, strange. Also i only see some very general info but no details. And i found a tab on the left where i assume the device should show up but there is a connection error. I also found an older version 1.x online from people complaining that it will not work any longer with the new complex passwords but looks the same for me… 
Maybe someone has an idea about this, or i can try to find an SMA forum what i could try else.

[image]
[image]
[image]

habuild, 2023-07-19:
So I think you are using the wrong password for SE, also the wrong user group. Normally you would use the USER usergroup.

[image]
This image should look like the red inverter below.
 You will probably see a padlock also 
[image]

[image]

When the icon is the red inverter icon, you will have a settings page to edit the passwords. In INSTALLER user both passwords can be edited here. 
[image]

I think it is complaining about the default password because you are trying to log in with 1111 when you have already changed to a different password. 
Try logging in with the USER account and user account password.  You might need to reset the passwords, the manual goes into that. The Question Mark ? bubble on the top right of sunny explorer has the manual on how to change the password.
Also try and avoid special characters in the password. Just go with lower/Upper case and numbers. Special chars can be a pain for SBFspot and this addon.

SBFspot uses the password from the USER account to connect and get the data

3gfisch, 2023-11-04:
Thanks, I tried all combinations i could, also with an old laptop and BT stick from a friend which works for his inverter. Either a password was set by someone and the tool is not reporting an error or there is an other problem with this inverter. Never got any data or lock symbol or real error message.. I will not invest more time in it but thanks a lot for the grate support!