habuild/hassio-addons #89: How to add triinv? Addon not showing.

herrtschaefer, 2024-01-03:
I added the repository to homeassistant but only see HAOS SBFSpot and MultiINV but nothing for the third Bluetooth inverter.  

what exactly does the MIS enable do?, do I need that when using the three addons?

habuild, 2024-01-03:
@herrtschaefer
You don't need multiple addons, just haos-sbfspot.

MIS-enabled is multi inverter setup, it sets up a little Bluetooth group between your inverters so that communicate between themselves and send out the info via a main inverter.

herrtschaefer, 2024-01-03:
Yea but i somehow only get data for one inverter, i tried setting up all inverters through the Bluetooth Mac Matt generator but i only get entities for one. Setting up multiple plants doesn’t solve the issue.

habuild, 2024-01-03:
Sorry I have no idea what you mean here "Bluetooth Mac Matt generator"

Have you setup the net-id for your inverters? you need to do this to get MIS to work.

Once MIS is working SBFspot will publish 3 MQTT data messages from the one addon. 
You will need to setup the MQTT sensors in home assistant. This addon can help with that but it is a bit fiddly.

issue #42 has some more user info in it. I haven't ever really found a definitive guide on how to set it up to be honest.

[This HA forum post might also help](https://community.home-assistant.io/t/sbfspot-bluetooth-homeassistant-addon/402671/322?u=hasqt)

closed 2024-01-05 by habuild