habuild/hassio-addons #92: Halt add-on with exit code 5

scauwe, 2024-01-22:
Hi all,

I'm trying to use the HAOS-SBFspot to read data from my Sunny Boy 3000TL-20 into HomeAssistant.
I'm trying to set it up to only integrate with HA, so no need for a Maria dB or PVoutput. I configured them with empty/fake values (I found in another ticket that this should do the trick).

Now, when starting, it did create the MQTT integration, but I do not get any devices or sensors.
Environment: HA on a RPi4.
HA Core version 2024.1.3

The startup logs of the add-on show the following:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sbfspot (no readiness notification)
s6-rc: info: service legacy-services successfully started
[00:00:08] INFO: 
[00:00:08] INFO:  Linux 6.1.63-haos-raspi #1 SMP PREEMPT Tue Jan  9 10:43:14 UTC 2024 aarch64
[00:00:08] INFO: 
|   INITIALISE MQTT SENSOR CREATION   |
sma
core-mosquitto
sma
homeassistant/sbfspot_{plantname}/sbfspot_{serial}
PrgVersion,Plantname,Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvSwVer,InvClass,InvType,InvTemperature,InvGridRelay,EToday,ETotal,GridFreq,PACTot,PAC1,UAC1,IAC1,OperTm,FeedTm,PDCTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2,BTSignal,InvWakeupTm,InvSleepTm
[00:00:09] INFO: Using Existing Device info

PrgVersion Plantname Timestamp SunRise SunSet InvSerial InvName InvTime InvStatus InvSwVer InvClass InvType InvTemperature InvGridRelay EToday ETotal GridFreq PACTot PAC1 UAC1 IAC1 OperTm FeedTm PDCTot UDC1 UDC2 IDC1 IDC2 PDC1 PDC2 BTSignal InvWakeupTm InvSleepTm
[00:00:10] INFO: Setting Up PrgVersion
[23:00:11] WARNING: Halt add-on with exit code 5
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```
The first startup showed something different, but I was to late to grab that.

In the home-assistant.log file, I found te following error:
`2024-01-22 23:50:46.480 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/a51a23d8_haos-sbfspot/stats - Container addon_a51a23d8_haos-sbfspot is not running
`

Thanks for your help

habuild, 2024-01-23:
Hi @scauwe 

I can't seem to find anything relating to error code 5 from s6 overlay or container errors. So I am not sure what that is pointing to.

It is stopping at the creation of the first MQTT sensor by the look of it.
A couple things to try.

A complete system reboot (not just restarting home assistant). That will clear out any ghost containers that might be related to the home assistant log error.

Then try starting the addon again and check the error. If it fails again, try starting the addon without the sensor creation option turned on.

scauwe, 2024-01-23:
Hi @habuild ,

Thanks for the answer. It guided me to what I needed.
The reboot did not help.
Disabling the MQTT topics showed that it was connecting to the inverter via bluetooth. When trying to publish the results, the following was logged:
```
Client null sending CONNECT
Connection error: Connection Refused: not authorised.
Error: The connection was refused.
Client null received CONNACK (5)
MQTT: Failed te execute '/usr/bin/mosquitto_pub' mosquitto client installed?
Error 1280 while publishing to MQTT Broker

```
In the mosquito help, I found that MQTT by default uses the HA users to authenticate. Since I was not using the default user, that was the problem! I created the user configured in HAOS-SBFspot in HA and restarted both MQTT and HAOS-SBFspot. SBFspot could now connect with te configured user and password without a problem. I reconfigured HAOS-SBFspot to create the topics and I found them as sensors in HA.

Thanks a lot!

closed 2024-01-23 by scauwe