habuild/hassio-addons #93: Sensor reading values not imported intoHA

scauwe, 2024-01-24:
Hi all,

After I got the HAOS-SBFspot connected to MQTT (ticket #92 ), I am now faced with the next problem:
all values are read from the inverter, but the publishing to the queue seems not to work. All MQTT sensor values are "Unknown".
The SBFspot does not show an error (except for the MySQL which I indeed did configure with dummy data).
In the debug info from MQTT, the HAOS-SBFspot entity, I do see the HAOS-SBFspot topics, but "most recently received messages" is "0" for all.

Relevant part of the log from HAOS-SBFspot:
```
[2024-01-24 14:20:38.124] Error: Can't open MySQL db [SBFspot] : Unknown server host 'core-mariadb' (-5)
MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}) "PrgVersion": "3.9.7","Plantname": "sma","Timestamp": "2024-01-24T14:20:38+0100","SunRise": "2024-01-24T08:30:00+0100","SunSet": "2024-01-24T17:16:00+0100","InvSerial": XXXX,"InvName": "SN: XXXX","InvTime": "2024-01-24T14:21:43+0100","InvStatus": "Ok","InvSwVer": "02.08.01.R","InvClass": "Solar Inverters","InvType": "SB 3000TL-20","InvTemperature": 36.020,"InvGridRelay": "Closed","EToday": 2.278,"ETotal": 40861.886,"GridFreq": 49.990,"PACTot": 598.000,"PAC1": 602.000,"UAC1": 228.020,"IAC1": 2.640,"OperTm": 53938.480,"FeedTm": 51766.522,"PDCTot": 630.000,"UDC1": 368.830,"UDC2": 0.000,"IDC1": 1.707,"IDC2": 0.000,"PDC1": 630.000,"PDC2": 0.000,"BTSignal": 73.333,"InvWakeupTm": "2024-01-24T08:48:49+0100","InvSleepTm": "2024-01-24T14:21:48+0100"
Client null sending CONNECT
Client null received CONNACK (0)
Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_sma/sbfspot_XXXX', ... (763 bytes))
Client null sending DISCONNECT
********************
* ArchiveDayData() *
********************
startTime: 24/01/2024 00:00:00
Wed Jan 24 14:20:39 2024: INFO: Done

```

Thanks,
Stefaan

habuild, 2024-01-24:
It sounds like your MQTT topics might not be matching.

```
homeassistant/sbfspot_sma/sbfspot_XXXX
```
That topic that the addon is posting to the MQTT server needs to be the same when Home Assistant is listening to the MQTT server.

**It is probably just a case of deleting the sensors from Home Assistant and re-adding with the addon.** They may have picked a different topic the first time they were created when the MQTT connection wasn't working in issue #92

You can use the listen to a topic in the MQTT integration to confirm it is being received in Home Assistant, then you will need to configure your sensors again.
```
Settings > Devices & Services > MQTT > configure > listen to a topic
```   
You could also use program like "MQTT explorer" to connect and troubleshoot, which is available as a windows download or a Home Assistant addon.

[MQTT explorer addon HA community forum](https://community.home-assistant.io/t/addon-mqtt-explorer-new-version/603739)

scauwe, 2024-01-26:
Hi all,

When removing all the sensors, HAOS-SBFspot adds them again on starting up. What I noticed is that these are added without the serial number:
[image]
Since these are created prior to connecting, I thought that this was normal.
I changed the MQTT topic configured in HAOS-SBFspot to include the serial number "hardcoded": Chagne from homeassistant/sbfspot_{plantname}/sbfspot_{serial} to homeassistant/sbfspot_{plantname}/sbfspot_2100XXXXXX
When deleting the sensors again, HAOS-SBFspot still generated config messages without the serial number.
So, next try was to remove the serial number in HAOS-SBFspot from the topic name:
homeassistant/sbfspot_{plantname}/sbfspot_
And now I do have results in the sensors.

So, no clue what the actual issue is right now. It seems logic that the serial number is unknown at the time of creation of the topics (since sbfspot was not yet started), but hardcoding the serial number in the topic did not help either.

Stefaan

habuild, 2024-01-26:
Run a purge sensors

Or at worst case copy your options (in yaml mode is easy for copy paste) and uninstall /reinstall the addon.

When the addon starts trying to create the sensors the first time it creates a file with the serial number it receives via mqtt from SBFspot. It gets abit stuck if it creates that file incorrectly.

You can't hard code the serial in the mqtt topic in options. That is is one of the sacrifices that had to be made to get the addon options to convert to something that sbfspot could use.

scauwe, 2024-01-27:
The purge sensors did not solve it.
Uninstall & reinstall did. Now I have the serial number as part of the topic.

Thanks!
Closing the issue.

closed 2024-01-27 by scauwe