habuild/hassio-addons #98: retreive history

TerrorSource, 2024-03-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Is it possible to add the parameters to the config in HA to retrieve the history?
it uses the following parameters in the original software:
-ad#                Set #days for archived daydata: 0-300
                    0=disabled, 1=today (default), ...
-am#                Set #months for archived monthdata: 0-300
                    0=disabled, 1=current month (default), ...
-ae#                Set #months for archived events: 0-300
                    0=disabled, 1=current month (default), ...

**Describe the solution you'd like**
have the parameters above as optional settings

**Describe alternatives you've considered**
...

**Additional context**
Optional settings might be the best option here but an short how-to to edit it via portainer is also fine.

habuild, 2024-03-30:
Hi I'm out of my country atm. So will be offering limited support for a few weeks.

In saying that.

You can log into the container using the homeassistant community ssh addon. You should be able to find the command in one of the other open issues or readme page(I think it's been a while since I looked)

Once you login to the container you can run SBFspot with any command line arguments you like. So you can run an extended archive that way.

Adding options to the command to offer a variable system for the archive would require a rewrite of this addon as the command is hard coded into a crontab in the dockerfile. 
This creates the crontab before the options are created and accessed from homeassistant. 

You can clone the addon and edit the crontab to suit your particular needs in the dockerfile if you like.

TerrorSource, 2024-04-02:
> Hi I'm out of my country atm. So will be offering limited support for a few weeks.
> 
> In saying that.
> 
> You can log into the container using the homeassistant community ssh addon. You should be able to find the command in one of the other open issues.
> 
> Once you login to the container you can run SBFspot with any command line arguments you like. So you can run an extended archive that way.
> 
> Adding options to the command to offer a variable system for the archive would require a rewrite of this addon as the command is hard coded into a crontab in the dockerfile. This creates the crontab before the options are created and accessed from homeassistant.
> 
> You can clone the addon and edit the crontab to suit your particular needs in the dockerfile if you like.

Thanks! No problem :)

I've run "./SBFspot -ad300 -am300 -ae300 and some other paratmeters via de commandline form within the docker itself. I used Portainer to get into the container.

closed 2024-04-02 by habuild