habuild/hassio-addons #99: No data on overview HAOS-SBFspot ( Help for noob )

thedudefxp, 2024-03-30:
Hello, i finally manage to install HAOS-SBFspot addon on home-assistant but i have no data in Overview. I have multiple error 21 on log but all other things seems fine. Pvoutput works and MariaDb too.
Im sorry if the question already ask, i'm pure noob, i checked the issue and google but cant find any help.

Thanks in advance for your help 

[image]

Here my log
`BT Signal=63.9%
Logon OK
SUSyID: 99 - SN: 2001207962
Device Name:      SN: 2001207962
Device Class:     Solar Inverters
Device Type:      SB 3000
Software Version: 12.12.210.R
Packet status: 21
SUSyID: 99 - SN: 2001207962
Device Status:      Ok
Packet status: 21
SUSyID: 99 - SN: 2001207962
Device Temperature: Information not available
Packet status: 21
getGridRelayStatus returned an error: 21
SUSyID: 99 - SN: 2001207962
Energy Production:
	EToday: 4.110kWh
	ETotal: 44910.312kWh
	Operation Time: 60173.11h
	Feed-In Time  : 54408.35h
Packet status: 21
getSpotDCPower returned an error: 21
Packet status: 21
getSpotACPower returned an error: 21
SUSyID: 99 - SN: 2001207962
DC Spot Data:
	MPPT 1 Pdc:   0.204kW - Udc: 394.00V - Idc:  0.520A
	MPPT 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	Calculated Total Pdc:   0.000kW
AC Spot Data:
	Phase 1 Pac :   0.178kW - Uac: 238.00V - Iac:  0.748A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.178kW - Calculated Pac:   0.178kW
	Efficiency  :    0.00%
SUSyID: 99 - SN: 2001207962
Grid Freq. : 49.98Hz
SUSyID: 99 - SN: 2001207962
Current Inverter Time: 2024-03-30T15:29:44+0100
Inverter Wake-Up Time: 2024-03-30T15:29:44+0100
Inverter Sleep Time  : 2024-03-30T15:29:45+0100
MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}) "PrgVersion": "3.9.7","Plantname": "SMA_SB_3000","Timestamp": "2024-03-30T15:30:03+0100","SunRise": "2024-03-30T06:17:00+0100","SunSet": "2024-03-30T19:05:00+0100","InvSerial": 2001207962,"InvName": "SN: 2001207962","InvTime": "2024-03-30T15:29:44+0100","InvStatus": "Ok","InvSwVer": "12.12.210.R","InvClass": "Solar Inverters","InvType": "SB 3000","InvTemperature": 0.000,"InvGridRelay": "?","EToday": 4.110,"ETotal": 44910.312,"GridFreq": 49.980,"PACTot": 178.000,"PAC1": 178.000,"UAC1": 238.000,"IAC1": 0.748,"OperTm": 60173.114,"FeedTm": 54408.355,"PDCTot": 0.000,"UDC1": 394.000,"UDC2": 0.000,"IDC1": 0.520,"IDC2": 0.000,"PDC1": 204.000,"PDC2": 0.000,"BTSignal": 63.922,"InvWakeupTm": "2024-03-30T15:29:44+0100","InvSleepTm": "2024-03-30T15:29:45+0100"
Client null sending CONNECT
Client null received CONNACK (0)
Client null sending PUBLISH (d0, q0, r0, m1, 'homeassistant/sbfspot_SMA_SB_3000/sbfspot_2001207962', ... (759 bytes))
Client null sending DISCONNECT`

habuild, 2024-03-30:
The error 21 is info not available I believe. That may resolve itself in time.

https://github.com/SBFspot/SBFspot/issues/514

As to the unknown mqtt sensors make sure the topic/message matches what is being sent from the addon to what is being received in homeassistant. You may need to delete and recreate the sensors. If you created the sensors before you had a connection to the inverter, it may be easier to  cut/copy your options and reinstall the addon and then create the sensors again.

thedudefxp, 2024-03-31:
Thanks you for you rreply
Sadly i already try to remove mqtt entity after purge, create and no again but it still Unknow everywhere.
Im gonna try to remove addon, db, mqtt and try a fresh install of all. i will come back here after that.
Sorry for that dumb question

habuild, 2024-03-31:
You shouldn't need to remove the DB, they aren't really linked. The DB and pvoutput are linked.

MQTT is mostly standalone.
You can check with mqtt Explorer to confirm the message is being sent, and to which topic. Also you need to make a user in the mqtt addon for sbfspot. You can't use the homeassistant user.

When you create the sensors the mqtt addon will list the topics that are being listened to.

We were all new once.
I'm overseas atm so a bit limited for connectivity.

thedudefxp, 2024-04-01:
Hello, i followed your advice. Deleted sbfspot, mqtt addons and mqtt integration. 
Rebooted HA 
Installed mqtt add user
Install and configure SBFspot
Start SBSspot with No - with Create and finally No
And after HA reboot TADAMMM :)
Thanks for help an old non english speaking noob like me, who discover HA two week ago only for your addon 
I forgot to gratz you for this nice addon and the work done ( my 14 years old inverter thanks you too, its now fully connected ).

Have à nice day and we can call this issue Closed

[image]

habuild, 2024-04-02:
:)

closed 2024-04-02 by habuild