habuild/hassio-addons #107: No data before 6 am

CHH25, 2024-06-18:
I have used HAOS-SBFspot since autumn 2023 and have now been aware that no data is registered before 6 am. I can see that cronjob daydata only runs between 6 am - 11pm, so this is as desgned.
However at summer time my SMA inverter generates energy already from about 4:30 am. 
I use ETotal as input to HA's Energy meter, and no energy data has been registered before 6 am in Solar Production.
Is it possible to configure starting time of the cronjob to run earlier, fx at 4 am?

CHH25, 2024-06-21:
Yes, the sum of generated energy before 6 am is registered at 6 am, and the following ETotal is correct.

I have installed HAOS-SBFspot on a RPi4, and I use Home Assistant OS 12.4, Core 2024.6.3. 

If I use your proposed SSH in my addon (Terminal & SSH) I get the following error: 

[image]
 
When I run your proposed service in Developer Tools/Services it seems to be ok. 
(this is the content in ‘home-assistant.log’):

2024-06-21 11:44:50.982 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Running websocket_api script
2024-06-21 11:44:50.982 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Executing step call service

However, I expected to see an update of ETotal, but I did not.

I have also added your proposed automation. It only triggers once an hour, and not every 5 minutes. 
(minutes: "5" needs to be replaced with minutes: /5).
Hence it only triggered at 05:05. The log content:

2024-06-21 05:05:00.481 DEBUG (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Automation trigger 'None' triggered by time pattern
2024-06-21 05:05:00.482 INFO (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Poll SMA inverter before 6 am: Running automation actions
2024-06-21 05:05:00.482 INFO (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Poll SMA inverter before 6 am: Executing step call service

It looks fine, but ETotal was not updated at 05:05 (see the graph below). 

[image]
 
In my research I can see that the service hassio.addon_stdin might have been deprecated. Can that be the reason for the problems?

habuild, 2024-06-21:
Not this addon
[image]

Needs to be this community one, which is also maintained by Frenck. You also need protection mode off I think. 
[Community SSH](https://github.com/hassio-addons/addon-ssh)
[image]

Yeah that automation should have been /5 minutes. It's not something I use, I just made it as an example for your feature request.

It might be a quotation error. It might also need to be a slightly different command/addon for the service quest.

It might need the quotes like this. You might have to play around with '' or double "" wrapping.

```
input: "/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq"
```

This is using the community addon ( a0d7b954_ssh <-- container id ), instead of trying to service call the SBFspot addon directly.
```
service: hassio.addon_stdin
data:
  addon: a0d7b954_ssh
  input: "docker exec -it addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"
```

[image]

It might need to route via the community addon, if it can't run directly as above.
[image]

You can also Fork this addon and adjust the docker file crontab to accommodate your feature request. 
To have it be dynamically changeable requires an overhaul and rewrite of this addon to bring inline with s6 overlay and redoing all the scripts etc.

habuild, 2024-06-22:
Actually it does look like _**hassio.addon_stdin**_ has been removed to close a security flaw. Or perhaps I couldn't get the quotes to line up.. which is always a possibility.

So that means you have to use the SSH method and a shell command. 

The shell command is VERY sensitive to quote wrapping. This is what worked for me. "double" arpund the whole command and 'single' around the sub command.
```
"docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"
``` 
To make the shell command in home assistant you can add this into your config. You need to restart HA when you make it or edit it
```
shell_command:
  sbfspot_update: >-
    ssh root@YourHostName.local -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa "docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"
```

For that command to work you need to setup the ssh as per [this guide ](https://community.home-assistant.io/t/sshing-from-a-command-line-sensor-or-shell-command/258731)

You can test command in the SSH community addon. either as
```
docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'
```

or as the full command once you have setup the SSH access and put your hostname or IP address in.

```
ssh root@YourHostName.local -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa "docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"
```

Once the shell command is working you can setup your automation to run the shell command, or you could make a dashboard button.

CHH25, 2024-06-22:
OK. Yesterday I really strugled with hassio.addon_stdin, but fine that you can confirm that the functionallity has been removed..

Today I have worked with your new proposal, using SSH method and a shell command. However with limited success.

I already had some key files, but have also generated new keys, and the ssh has been setup according to your guide lines.

When I start 'Advanced SSH & Web Terminal' I can succesfully run this ssh:

`ssh claus@192.168.1.35 -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa `

Next I can run this succesfully:

`docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'`

This updates ETotal as wanted (not visible in the picture below).

[image]

However, if I combine these two commands as you have sugested, I get the error : Permission denied. 
I have tested several combinations of quotes, but with the same error.

`ssh claus@192.168.1.35 -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa "docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"`

[image]

Today I have used hours trying to solve this, but without any luck. It seems to be a problem handling the id keys, but I have not been able to find a solution. I hope you can help me.

habuild, 2024-06-23:
You need to setup the keys not for your login, but for root login.

so this command
```
ssh claus@192.168.1.35 -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa 
```

should be _root@192.168.1.35_ etc etc etc

```
ssh root@192.168.1.35 -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa "docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"
```

You need to run the command as a root user in the sbfspot container.

CHH25, 2024-06-24:
Thanks for your help.

The scripts have been running this morning, and now I get the meterings from my inverters before 6 am.

habuild, 2024-06-26:
The 5:50 retrieves data?
It looks like Bluetooth is failing at 5:51. Is there something else grabbing BT services or the inverters at that time? 

It seems odd that 5:55 would fail. The 6am trigger from the addon should be happening at 6am and is relatively instantaneous.

Two Bluetooth calls to the inverters will clash. 5:55 and 6am shouldn't be clashing though.

You could try running stopping the /5 trigger at 5:50. Then try a single automation for 5:54 or 5:56 and see if they clash.

CHH25, 2024-06-26:
Yes, all data before 5:55 are retrieved ok. The problem only occurs at 5:55.

I will try stopping the /5 trigger at 5:50, and add a new automation that triggers at 5:54.

CHH25, 2024-06-30:
I still have a problem to get data from 05:55.

During the last days I have made several tests (however, only one test is possible each day).

Test 1 (27/6-2024):
Stopped triggering at 05:50, and made an automation that triggered at 05:54. Ok

Test 2 (28/6-2024):
Running the original /5 automation that stops triggering at 06:00. Test failed at 05:55

Test 3 (29/6-2024): 
Logging level changed from Info to Debug.
Stopped triggering at 05:50, and made an automation that triggered at 05:56. Ok

Test 4 (30/6-2024):
Running the original /5 automation that stops triggering at 06:00. However this time I started triggering at 03:00. Test failed only at 05:55.

Here is the log from the 4 tests. However I have only included data related to the automation 'poll_sma_inverter_before_6_am', 'poll_inverters_at_5_55', and bluetooth events.

```
2024-06-27 05:50:00.084 DEBUG (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Automation trigger 'None' triggered by time pattern
2024-06-27 05:50:00.085 INFO (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Poll SMA inverter before 6 am: Running automation actions
2024-06-27 05:50:00.085 INFO (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Poll SMA inverter before 6 am: Executing step call service

2024-06-27 05:52:16.762 INFO (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Bluetooth scanner has gone quiet for 90s, restarting

2024-06-27 05:52:34.465 WARNING (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Successful fall-back to passive scanning mode after active scanning failed (4/4)

2024-06-27 05:54:00.315 DEBUG (MainThread) [homeassistant.components.automation.poll_inverters_at_5_55] Automation trigger 'None' triggered by time
2024-06-27 05:54:00.316 INFO (MainThread) [homeassistant.components.automation.poll_inverters_at_5_55] Poll inverters at 5.55: Running automation actions
2024-06-27 05:54:00.317 INFO (MainThread) [homeassistant.components.automation.poll_inverters_at_5_55] Poll inverters at 5.55: Executing step call service

2024-06-27 05:55:00.084 DEBUG (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Automation trigger 'None' triggered by time pattern
2024-06-27 05:55:00.085 DEBUG (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Conditions not met, aborting automation. Condition summary: {'trigger/0': deque([{'path': 'trigger/0', 'timestamp': datetime.datetime(2024, 6, 27, 3, 55, 0, 85052, tzinfo=datetime.timezone.utc), 'changed_variables': {'this': {'entity_id': 'automation.poll_sma_inverter_before_6_am', 'state': 'on', 'attributes': {'id': '1718900889453', 'last_triggered': datetime.datetime(2024, 6, 27, 3, 50, 0, 85505, tzinfo=datetime.timezone.utc), 'mode': 'single', 'current': 0, 'friendly_name': 'Poll SMA inverter before 6 am'}, 'last_changed': '2024-06-26T21:49:15.976709+00:00', 'last_reported': '2024-06-27T03:50:06.332132+00:00', 'last_updated': '2024-06-27T03:50:06.332132+00:00', 'context': {'id': '01J1BX15QWJARG1Z7SF129FC2N', 'parent_id': None, 'user_id': None}}, 'trigger': {'id': '0', 'idx': '0', 'alias': None, 'platform': 'time_pattern', 'now': datetime.datetime(2024, 6, 27, 5, 55, 0, 84213, tzinfo=zoneinfo.ZoneInfo(key='Europe/Copenhagen')), 'description': 'time pattern'}}}]), 'condition/0': deque([{'path': 'condition/0', 'timestamp': datetime.datetime(2024, 6, 27, 3, 55, 0, 85126, tzinfo=datetime.timezone.utc), 'result': {'wanted_time_after': datetime.datetime(2024, 6, 27, 2, 26, 23, 166628, tzinfo=<UTC>), 'result': True}}]), 'condition/1': deque([{'path': 'condition/1', 'timestamp': datetime.datetime(2024, 6, 27, 3, 55, 0, 85832, tzinfo=datetime.timezone.utc), 'result': {'after': datetime.time(0, 0), 'now_time': datetime.time(5, 55, 0, 85871), 'before': datetime.time(5, 52), 'result': False}}])}

2024-06-27 05:57:16.838 INFO (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Bluetooth scanner has gone quiet for 90s, restarting

2024-06-27 05:57:34.515 WARNING (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Successful fall-back to passive scanning mode after active scanning failed (4/4)

2024-06-28 05:50:00.314 DEBUG (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Automation trigger 'None' triggered by time pattern
2024-06-28 05:50:00.316 INFO (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Poll SMA inverter before 6 am: Running automation actions
2024-06-28 05:50:00.317 INFO (MainThread) [homeassistant.components.automation.poll_sma_inverter_before_6_am] Poll SMA inverter before 6 am: Executing step call service

2024-06-28 05:51:54.950 INFO (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Bluetooth scanner has gone quiet for 90s, restarting

2024-06-28 05:52:12.614 WARNING (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Successful fall-back to passive scanning mode after active scanning failed (4/4)

2024-06-28 05:53:54.960 INFO (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Bluetooth scanner has gone quiet for 90s, restarting

2024-06-28 05:54:12.630 WARNING (MainThread) [habluetooth.scanner] hci0 (D8:3A:DD:38:12:3A): Successful fall-back to passive scanning mode after active scanning failed (4/4)

[... cut ...] 8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"`, return code: 255
NoneType: None
```

My observation:
When the /5 automation fails (fx 2024-06-30 05:55:10.564) the return code from the shell_command is: 255. Otherwise the return code is: 0. However I do not know what return code = 255 means.

My next test tomorrow will be to test a workaround. Stopping the /5 automation at 05:54 and add a new automation that triggers only at 05:55 (simalar to test 1 and test 3).

habuild, 2024-06-30:
> 2024-06-28 05:55:10.416 ERROR (MainThread) [homeassistant.components.shell_command] Error running command: `ssh ***@***.*** -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa "docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"`, return code: 255
> 2024-06-30 05:55:10.564 ERROR (MainThread) [homeassistant.components.shell_command] Error running command: `ssh ***@***.*** -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa "docker exec addon_a51a23d8_haos-sbfspot bash -c '/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"`, return code: 255

That connection at 5:55 is being blocked. Something else is connected to
the inverter at that. SBFspot shouldn't be trying to connect until 6am so I
don't believe it is that, that is blocking the connection. Is there
something else you log the data with or is the time out by 5 mins on the
inverter?

```
2024-06-30 05:55:10.564 DEBUG (MainThread)
[homeassistant.components.shell_command] Stdout of command: `ssh
***@***.*** -o UserKnownHostsFile=/config/.ssh/known_hosts -i
/config/.ssh/id_rsa "docker exec addon_a51a23d8_haos-sbfspot bash -c
'/usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -finq'"`, return code:
255:
b"SBFspot V3.9.9\nYet another tool to read power production of SMA solar
inverters\n(c) 2012-2024, SBF (https://github.com/SBFspot/SBFspot)\nCompiled
for Linux (LE) 64 bit with MySQL support\nCommandline Args: -v -ad1 -am0
-ae0 -mqtt -finq\nReading config '/usr/bin/sbfspot/SBFspot.cfg'\nSun Jun 30
05:55:01 2024: INFO: Starting...\nsunrise: 04:31\nsunset :
21:57\nConnecting to 00:80:25:a5:38:1a (1/10)\nConnecting to
00:80:25:a5:38:1a (2/10)\nConnecting to 00:80:25:a5:38:1a
(3/10)\nConnecting to 00:80:25:a5:38:1a (4/10)\nConnecting to
00:80:25:a5:38:1a (5/10)\nConnecting to 00:80:25:a5:38:1a
(6/10)\nConnecting to 00:80:25:a5:38:1a (7/10)\nConnecting to
00:80:25:a5:38:1a (8/10)\nConnecting to 00:80:25:a5:38:1a
(9/10)\nConnecting to 00:80:25:a5:38:1a (10/10)\nSun Jun 30 05:55:10 2024:
CRITICAL: bthConnect() returned -1\nSun Jun 30 05:55:10 2024: INFO: Done.\n"
2024-06-30 05:55:10.564 ERROR (MainThread)
```

CHH25, 2024-06-30:
No, I do not have other devices that connect to the inverters in the morning hours except my Home Assistant with SBFspot. I sometimes use Sunny Explorer from my pc, but the pc is always shut down during the night.

I think that the inverter time is correct. Fx the event from 2024-06-30 05:50:07.781 contains 'Current Inverter Time' for both of my inverters. The difference between the inverters is only 3 seconds.

Furthermore this information from Sunny Explorer:

[image]

[image]

I have tried to find information about event 10223, but I do not think it is relevant here. Furthermore this time is different every day.

Do you have any suggestions to retrieve relevant information, using Sunny Explorer?

habuild, 2024-06-30:
Ugh my apologies I just noticed the Archive poll at 5:55am. That will be what is blocking your automation. I skipped over the time when looking at the docker file via mobile phone. I will add a -mqtt call to that. Then you should only need to run your automation until 5:50.

CHH25, 2024-06-30:
Thanks for your explanation. Nice, if your suggested correction will solve the problem.

habuild, 2024-06-30:
Should be live now to install and test for next 5:55am.

CHH25, 2024-07-01:
Thanks. It worked fine this morning.

MmeHulot, 2025-03-22:
i edited crontab as you described above  [](https://github.com/habuild/hassio-addons/issues/107#issuecomment-2178794962)   and added an -finq option:

`@reboot sleep 30 && /usr/bin/sbfspot/SBFspot -v -ad0 -am0 -mqtt -finq > /dev/stdout
*/5 6-22 * * *    /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt **-finq** > /dev/stdout
55 05 * * *       /usr/bin/sbfspot/SBFspot -v -sp0 -ad14 -am1 -ae1 -mqtt -finq > /dev/stdout
@reboot sleep 60 && /usr/bin/sbfspot/SBFspotUploadDaemon -c /usr/bin/sbfspot/SBFspotUpload.cfg > /dev/stdout
*/5 6-22 * * * /usr/bin/sbfspot/taillog.sh`
but the changes were not persistent. After restarting the Addon the `-finq` option disappeard

habuild, 2025-03-22:
@MmeHulot 

Sorry your question is unclear, please open a new issue and detail your query.

closed 2025-03-22 by habuild