habuild/hassio-addons #117: Addon stop working with s6-rc-init error

vitalisator, 2024-12-20:
**Describe the bug**
All works fine before. Since some days addin stops working with error:
s6-rc-init: fatal: unable to supervise service directories in /run/s6-rc/servicedirs: Broken pipe
s6-rc: fatal: unable to take locks: No such file or directory
s6-linux-init-shutdownd: warning: /run/s6/basedir/scripts/rc.shutdown exited 111

HAOS-SBFspot Version: 2024.7.1
Container Name: a51a23d8-haos-sbfspot

Home Assistant

    Core 2024.12.4
    Supervisor 2024.12.0
    Operating System 14.1
    Frontend 20241127.8

Probably it is broken by some HA Updates. 
Reinstalling of HAOS-SBFspot addin do not resolve the problem.

vitalisator, 2024-12-20:
There is nothing special with my HA Installation on Raspberry Pi 5. Just one thing may be worth of mentioning. I'm using an additional SSD on mountpoint /share to expanding the storage space for the immich AddOn (replacement for google Photos). 
`/dev/nvme0n1p1 on /mnt/data/supervisor/share type ext4 (rw,relatime)`
But I do not see how this could impact the HAOS-SBFspot

I'm try to dig into the root cause and installed the portainer AddIn, so I can get into CLI. Here I see the SBFspot files:
```
a51a23d8-haos-sbfspot:/usr/bin/sbfspot# ls -lh
total 3M     
-rwxr-xr-x    1 root     root      450.1K Jun 30 23:27 SBFspot
-rwxr-xr-x    1 root     root        9.4K Jun 30 23:20 SBFspot.default.cfg
-rwxr-xr-x    1 root     root        1.8K Jun 30 23:20 SBFspotUpload.default.cfg
-rwxr-xr-x    1 root     root      130.1K Jun 30 23:30 SBFspotUploadDaemon
-rwxr-xr-x    1 root     root      442.6K Jun 30 23:21 TagListDE-DE.txt
-rwxr-xr-x    1 root     root      429.1K Jun 30 23:21 TagListEN-US.txt
-rwxr-xr-x    1 root     root      462.3K Jun 30 23:21 TagListES-ES.txt
-rwxr-xr-x    1 root     root      471.5K Jun 30 23:21 TagListFR-FR.txt
-rwxr-xr-x    1 root     root      451.3K Jun 30 23:21 TagListIT-IT.txt
-rwxr-xr-x    1 root     root      434.7K Jun 30 23:21 TagListNL-NL.txt
-rwxr-xr-x    1 root     root       47.4K Jun 30 23:21 date_time_zonespec.csv
-rwxr-xr-x    1 root     root       11.9K Jun 30 23:20 genBluetoothConfig.sh
-rwxr-xr-x    1 root     root       11.9K Jun 30 23:20 genEthernetConfig.sh
-rwxr-xr-x    1 root     root        9.6K Jun 30 23:20 genSenBlue
-rwxr-xr-x    1 root     root        9.7K Jun 30 23:20 genSenEth
-rwxr-xr-x    1 root     root       12.0K Jun 30 23:20 generateConfig.sh
-rwxr-xr-x    1 root     root       55.7K Jun 30 23:20 mqttSensorConfig
-rwxr-xr-x    1 root     root       18.4K Jun 30 23:20 purgeSen
-rwxr-xr-x    1 root     root         906 Jun 30 23:20 taillog.sh
```

Can you explain the location (absence) of the SBFspot.cfg? I assume that is generated from the UI Config and stored somewhere outside of the container, but where?

If a look at the running processes, there is no evidence that SBFspot is running, it seems that the S6 stucks on startup and then ends in shutdown or whatever state.
```
a51a23d8-haos-sbfspot:/usr/bin/sbfspot# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdownd -d3 -c /run/s6/basedir -g 3000 -C -B
   25 root      0:00 bash
   61 root      0:00 bash
   68 root      0:00 less SBFspot.default.cfg
   69 root      0:00 bash
   78 root      0:00 ps

```
Can you suggest where I can look for the logs or errors?

thank you for the support!

habuild, 2024-12-21:
Yeah S6 is stopping the addon before it starts, so it won't be active. Which is also why the config files don't exist, they are created when the addon starts under homeassistant by S6 scripts. 

`docker exec -it $(docker ps -f name=haos-sbfspot -q) /bin/bash` this command allows you to log into the container when it is running from [Advanced SSH & Web Terminal](https://github.com/hassio-addons/addon-ssh)(or Putty although I think you need special access for that) instead of portainer. just FYI

Did you start the container via portainer to log in to see the files? In theory you shouldn't be able to see the files if the addon isn't running (although I may be forgetting a feature of portainer when saying that).

`/usr/bin/sbfspot/genBluetoothConfig.sh /usr/bin/sbfspot/SBFspot.cfg /usr/bin/sbfspot/SBFspotUpload.cfg`
This command should generate your configs if you want to try and create them from within the addon with portainer instead of S6.

Although I tried starting it under portainer and it exits fairly quickly.

All that is to say, if you have it running long enough to log in to see the files, has it stopped stopping in home assistant?

If there is nothing obvious in Home assistant general or supervisor logs, I am not sure where to start. This is not an error I have encountered. I also agree I don't think the extra SSD should be a problem( I assume that existed when SBFspot previously worked.)

Any other significant changes since the addon last worked? Any other addons installed that may changed something?
You could try a fresh install on a separate SD/SSD or a VM and see if that still gives you an issue for testing purposes.

vitalisator, 2024-12-21:
The S6 definitely seems to have a problem, but the add-in won't close. Rather, the following is logged over and over again:
```
s6-rc-init: fatal: unable to supervise service directories in /run/s6-rc/servicedirs: Broken pipe
s6-rc: fatal: unable to take locks: No such file or directory
s6-linux-init-shutdownd: warning: /run/s6/basedir/scripts/rc.shutdown exited 111
s6-rc-init: fatal: unable to supervise service directories in /run/s6-rc/servicedirs: Broken pipe
s6-rc: fatal: unable to take locks: No such file or directory
s6-linux-init-shutdownd: warning: /run/s6/basedir/scripts/rc.shutdown exited 111
s6-rc-init: fatal: unable to supervise service directories in /run/s6-rc/servicedirs: Broken pipe
...
```
And before you ask, the Watchdog in the Add-On settings is deactivated.
I was also able to login via CLI and generate configs as you mentioned. I can also run the SBFspot binary and see the values fetched via Bluetooth in STDOUT vith -v2 Parameter.

So I think the only Problem is here that S6 environment is some kind broken for me.
I never worked with S6 before, so I need first to understand the basic mechanics.

habuild, 2024-12-21:
[s6 Overlay github](https://github.com/just-containers/s6-overlay)
The s6 docs start there.

The weird thing is I haven't had anyone else report this yet. I haven't changed the addon or moved to the latest base container(which could possibly update s6) or similar so I am not sure why it would have decided to change for you. Homeassistant or HAOS versions shouldn't affect the container either.

vitalisator, 2024-12-22:
while reading the docs and googling around I noticed that /run/s6-rc/servicedirs is something internal and should be present even the s6 crashes. https://skarnet.org/lists/supervision/2721.html
On my instance exactly that is a problem, the whole /run/s6-rc directory is missing after the s6 crashes. I can also reproduce that with fresh re-installation of Add-On and just start it with my Config, that worked fine before.

That points me to a suspiction. I had recently an update to HAOS 14.1 on Raspberry PI 5, is it possible that this special Update (I think the RP 5 is an different image that your VM one) has something to do with s6 incompatibilities on the Add-On?

I have actually not jet checked if it's possible to rollback the 14.1 to an earlier version.

habuild, 2024-12-22:
I had read that page myself, being a bug from 2020 it doesn't really seem relevant, other than pointing out the existence of the files. The addon isn't set to initiate these things specifically. As I don't get it to call s6 commands beyond what it automatically does. So I would think it should fail across all systems(which it hasn't so far). 

It seems very odd that it would pi5 specifically related. Although I don't recall helping others who are running it on pi5(There is possibly others who also didn't need help, I am only basing that off of support requests here or HA forums)

This does make me wonder if there is an apparmor permissions requirement that might be different for the pi5. Although I think that would again cause problems for a pi4 also.. 
https://github.com/oxyde42/hassio-addons/blob/main/haos-sbfspot/apparmor.txt
It looks like that is allowed by this file, perhaps apparmor is blocking it elsewhere..

vitalisator, 2024-12-22:
any chance to deactivate apparmor for testing?

habuild, 2024-12-22:
I don't think you can easily disable apparmor in home assistant HAOS. It is pretty locked down by default. That file I linked previously is the just the profile that gets loaded by HAOS.

habuild, 2024-12-23:
https://github.com/habuild/hassio
try ether-sbfspot from my test repo

vitalisator, 2024-12-23:
> https://github.com/habuild/hassio try ether-sbfspot from my test repo

this one is working for me. If I understand it correct, that is without apparmor policy at all, right?
If so, the main question is, what is changed in PI 5 latest release :-)

habuild, 2024-12-23:
Yeah that is odd that there would be a specific change to pi5. 
I'm gonna update ether-sbfspot to enable the apparmor profile again, so you can confirm this is the issue. It's looking like a permission issue for that file it's trying to reach, although I looks like the wildcards should be covering that..

anyway let me know if the problem reoccurs with app armor.

vitalisator, 2024-12-23:
interesting, this one is also works for me.
I will provide the docker inspect output from working ether-sbfspot and not working sbfspot, maybe you have another idea what is different.
[inspect_ether-sbfspot.txt](https://github.com/user-attachments/files/18228606/inspect_ether-sbfspot.txt)
[inspect_sbfspot.txt](https://github.com/user-attachments/files/18228660/inspect_sbfspot.txt)

If I do simple diff, the only difference seems to be container image version 2024.7.1 vs 2024.12.2
Also has the haos-sbfspot apparmor.txt an additional "network" capability, but I'm not sure if that matters at any.

habuild, 2024-12-23:
That is odd. They are basically identical, although ether-spot has later packages, but that is purely because they were needed to be updated to recompile the addon.

vitalisator, 2024-12-23:
just a side note. I could be fine with ether-sbfspot but unfortunately it has another issue:
```
MQTT: Publishing (homeassistant/sbfspot_{plantname}/sbfspot_{serial}),"PrgVersion": "3.9.11","Plantname": "private","Timestamp": "2024-12-23T12:20:49+0100","SunRise": "2024-12-23T08:17:00+0100","SunSet": "2024-12-23T15:54:00+0100","InvSerial": 2100333XXX,"InvName": "SN: 2100333XXX","InvTime": "2024-12-23T12:20:54+0100","InvStatus": "Ok","InvSwVer": "03.01.05.R","InvClass": "Solar Inverters","InvType": "SB 5000TL-20","InvTemperature": 52.630,"InvGridRelay": "Closed","EToday": 1.197,"ETotal": 61085.156,"GridFreq": 50.000,"PACTot": 610.000,"PAC1": 610.000,"UAC1": 239.160,"IAC1": 2.555,"OperTm": 57593.044,"FeedTm": 55811.003,"PDCTot": 620.000,"UDC1": 347.320,"UDC2": 290.010,"IDC1": 1.020,"IDC2": 0.918,"PDC1": 354.000,"PDC2": 266.000,"BTSignal": 74.510,"InvWakeupTm": "2024-12-23T08:53:37+0100","InvSleepTm": "2024-12-23T12:20:55+0100"
Unable to connect (Lookup error.).
MQTT: Failed to execute '/usr/bin/mosquitto_pub' mosquitto client installed?
Error 256 while publishing to MQTT Broker
```

habuild, 2024-12-23:
Alright give ether-sbfspot 2024.12.5 a go

vitalisator, 2024-12-23:
Happy with new Version, thanks a lot!

closed 2024-12-23 by vitalisator