habuild/hassio-addons #121: Read solar data every minute

henkpls, 2025-02-27:
I would like to have the data update every minute, but only for HA. So I tested this by adding the following lines to the cron file:

1-59/5 6-22 * * *    /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -nosql > /dev/stdout
2-59/5 6-22 * * *    /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -nosql > /dev/stdout
3-59/5 6-22 * * *    /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -nosql > /dev/stdout
4-59/5 6-22 * * *    /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt -nosql > /dev/stdout

It works great, I get an update every minute. Can this be implemented?

habuild, 2025-02-27:
[https://crontab.guru/#1-59/5_6-22_*_*_*](https://crontab.guru/#1-59/5_6-22_*_*_*)
“At every 5th minute from 1 through 59 past every hour from 6 through 22.”

[https://crontab.guru/#2-59/5_6-22_*_*_*](https://crontab.guru/#2-59/5_6-22_*_*_*)
“At every 5th minute from 2 through 59 past every hour from 6 through 22.”

[https://crontab.guru/#3-59/5_6-22_*_*_*](https://crontab.guru/#3-59/5_6-22_*_*_*)
“At every 5th minute from 3 through 59 past every hour from 6 through 22.”

[https://crontab.guru/#4-59/5_6-22_*_*_*](https://crontab.guru/#4-59/5_6-22_*_*_*)
“At every 5th minute from 4 through 59 past every hour from 6 through 22.”

I believe this is what you are trying to achieve

[https://crontab.guru/#1-59_6-22_*_*_*](https://crontab.guru/#1-59_6-22_*_*_*)

“At every minute from 1 through 59 past every hour from 6 through 22.”

--------------------------------------------------------------------

So you would want to add the line as such
```
1-59 6-22 * * * /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt > /dev/stdout
```

or more completely at [Line 91 in the dockerfile](https://github.com/habuild/hassio-addons/blob/main/haos-sbfspot/Dockerfile#L91C1-L91C114)

```
RUN echo '*/5 6-22 * * *    /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt > /dev/stdout' >> /etc/crontabs/root

needs to change to 

RUN echo '1-59 6-22 * * *    /usr/bin/sbfspot/SBFspot -v -ad1 -am0 -ae0 -mqtt > /dev/stdout' >> /etc/crontabs/root
```

Then you just need to fork my repo and change the dockerfile to incorporate the line above.
You then just need to change a couple of lines in the dockerfile and config.yaml to put to your own repo and and update version number.

You also need the .github [https://github.com/habuild/hassio-addons/tree/main/.github](https://github.com/habuild/hassio-addons/tree/main/.githubhttps://github.com/habuild/hassio-addons/tree/main/.github) directory to compile the addon. You have to manually copy that I think. It won't exist in your fork.

You can compare versions here between this repo and my other test repo for lines you need to change in dockerfile and config.yaml. It just needs to point to your repo address

[Test version config.yaml](https://github.com/habuild/hassio/blob/main/haos-sbfspot/config.yaml)

[Main version config.yaml](https://github.com/habuild/hassio-addons/blob/main/haos-sbfspot/config.yaml)

[Test version dockerfile](https://github.com/habuild/hassio/blob/main/haos-sbfspot/Dockerfile)

[Main version dockerfile](https://github.com/habuild/hassio-addons/blob/main/haos-sbfspot/Dockerfile)

henkpls, 2025-02-28:
Thank you for the reply and the help. I used the cron lines above because I wanted to keep the old functionality (every 5 minutes) and add a **minimal** update for the minutes between with **nosql** option to reduce performance impact. Also I'm not sure if a DB update every minute doesn't impact the upload to PVoutput. As this is a minimal update I hoped it could be implemented without forking :)

habuild, 2025-02-28:
The problem is the **minimal** change you are introducing is a **clash** of timing that **interferes** with the normal execution of the command at the 5th minute. 

Your alternate to forking the project to achieve your desire is using an automation from within home assistant

such as 

```
description: ""
mode: single
triggers:
  - trigger: time_pattern
    minutes: 1-4,6-9,11-14,16-19,21-24,26-29,31-34,36-39,41-44,46-49,51-54,56-59 6-22
conditions:
  - condition: state
    entity_id: binary_sensor.haos_sbfspot_running
    state: ""
actions:
  - action: shell_command.sbfpot_finq
    data: {}
```

with the shell command being

```
shell_command:
  # poll SBFspot Inverter with -finq
  sbfpot_finq: ssh root@homeassistant.local -o UserKnownHostsFile=/config/.ssh/known_hosts -i /config/.ssh/id_rsa 'docker exec docker exec addon_a51a23d8_haos-sbfspot /SBFspot -v -finq -mqtt'
```

You need to setup the ssh keys to achieve this.
There is an SSH guide [here](https://community.home-assistant.io/t/sshing-from-a-command-line-sensor-or-shell-command/258731)

closed 2025-02-28 by habuild